PX4/PX4-Autopilot #19612: GPS not found error when using PX4 as UAVCAN/Cyphal Node

truckerDrew, 2022-05-08:
When attempting to use PX4 with the [NXP S32K146 evaluation board](https://www.nxp.com/design/development-boards/automotive-development-platforms/s32k-mcu-platforms/can-fd-development-system-for-drones-rovers-and-mobile-robotics:UCANS32K146) with the instructions from their gitbook, [here](https://nxp.gitbook.io/ucans32k1/px4-autopilot/building-and-flashing-px4), there is an error thrown ostensibly from [gps.cpp](https://github.com/PX4/PX4-Autopilot/blob/ce4e9f66908afe7232b139ee15aeb847e3895cd4/src/drivers/gps/gps.cpp#L795).  I am using Ubuntu 20.04 LTS.

The intended use for the eval board is to communicate PWM to an ESC via UAVCAN/Cyphal.  Currently, when a serial connection is established, the gps error "ERROR [gps] failed to open /dev/ttyS1 err: 1" is repeatedly thrown in an infinite loop.  The nuttx/px4 shell is available while the errors are being thrown.  

What is the best course of action for disabling GPS checks with PX4 when the firmware is used for mavlink/uavcan compatibility and not as the main FMU CAN node?

Alternatively, would one of the other options listed on the NXP gitbook make more sense for the purposes of setting up a CAN node?

PetervdPerk-NXP, 2022-05-10:
We found a regression in the gps code causing the UART stop working.
For now the workaround would be removing the 921600 entry in the folowing table.
https://github.com/PX4/PX4-GPSDrivers/blob/master/src/ubx.cpp#L110

We're looking for a more permanent fix.

dagar, 2022-05-10:
Any idea what the root cause is?

PR #19637 (2022-05-11): Update UCANS32K146 clocking to use XTAL and support higher pheriphal …
Use XTAL/SPLL instead of SIRC to have a higher peripheral frequencies to support higher data rates.

Should fix #19612 which caused that the UCANS32K146 UART driver would fail when requested a high baud rate.