PX4/PX4-Autopilot #19617: Gradle assemble went wrong when install Fast-RTPS-Gen

cjychenjiayi, 2022-05-09:
[image]
this is what shows in my terminal. My gradle version is 6.3
[image]
this is the corresponding build.gradle document, the blue part is where the error happened I don't know why it worked improperly, I just done the same thing correctly at another computer.  I followed the tutorial but stuck at "gradle assemble" step.
Looking forward to help, thx

husseinalijaafar, 2022-05-09:
How did you install gradle? sdkman?

cjychenjiayi, 2022-05-10:
> How did you install gradle? sdkman?

yeah, by sdkman

closed 2022-05-13 by cjychenjiayi