PX4/PX4-Autopilot #19626: RTT too high for timesync

Abner0907, 2022-05-10:
hi, I have a question, in which situation, it will print 
>[mavlink] [timesync] RTT too high for timesync: 165 ms (sender: 1)

My print information shows as below:

[image]
 I have seen @bkueng  had add a print information https://github.com/PX4/PX4-Autopilot/pull/15539, I am very appreciate anyone who can leave some notes here.

Abner0907, 2022-05-11:
This is my test log, https://logs.px4.io/plot_app?log=3a7a7624-41c0-4f1d-b450-8475744ed50d

bkueng, 2022-05-11:
compid=1 is the autopilot, do you have a routing loop?

Abner0907, 2022-05-11:
@bkueng I think I  do not have a routing loop in autopilot.  But I have a onboard computer and send the mavlink to fc(autopilot) through serial. I have checked, it do not send the message id 'MAVLINK_MSG_ID_TIMESYNC' to the fc.

Abner0907, 2022-05-11:
@bkueng 
Thanks,  I get the cause reason.
I have add an outside mavlink `handle_message_odometry()` which will sync time below, it cause the issues.
https://github.com/PX4/PX4-Autopilot/blob/fc7e979d84715ca71e9de025492a1c59673e415d/src/modules/mavlink/mavlink_receiver.cpp#L1397

closed 2022-05-12 by Abner0907