PX4/PX4-Autopilot #19667: Preflight Fail in HITL with v1.13.0

Katawann, 2022-05-17:
**Describe the bug**
When I run PX4 v1.13.0-beta on Pixhawk 4 in HITL with jmavsim, it fails to arm because of `Preflight Fail` (compass, accel, gyro and baro). I tried with 3  different Pixhawk 4 but the issue is always the same

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the [steps](https://docs.px4.io/v1.12/en/simulation/hitl.html) to have PX4 in HITL mode
2. It fails with:
```
$ ./Tools/jmavsim_run.sh -q -s -d /dev/ttyACM0 -b 921600 -r 250
Buildfile: /home/PX4-Autopilot/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
MSG: Preflight Fail: Compass Sensor 0 missing
MSG: Preflight Fail: Accel Sensor 0 missing
MSG: Preflight Fail: Gyro Sensor 0 missing
MSG: Preflight Fail: Baro Sensor 0 missing
Init MAVLink
MSG: Failsafe mode deactivated	
MSG: Arming denied! Flight termination active
```

**Expected behavior**
Sensors should not be failing as we are do a simulation through jMAVSim

julianoes, 2022-05-20:
I can see this issue in Gazebo as well, even without the `Sensor 0 missing` error:

```
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
[hardfault_log] Fault Log is Armed

FW git-hash: 887fe7dba2d2d0dbdf8a2b9141f1b0de781ebc4e
FW version: 1.13.0 0 (17629184)
FW git-branch: master
OS: NuttX
OS version: Release 10.1.0 (167837951)
OS git-hash: c5c7d2b4f26f52f1dfb425ebde83328606b65d4f
Build datetime: May 20 2022 13:28:32
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000200000000353436333138511900200033
MCU: STM32F76xxx, rev. Z
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
INFO  [parameters] BSON document size 262 bytes, decoded 262 bytes (INT32:9, FLOAT:4)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
Loading airframe: /etc/init.d/airframes/1001_rc_quad_x.hil
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [px4io] IO FW CRC match
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix appended to /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
ekf2 [444:237]
Starting MAVLink on /dev/ttyS1
WARN  [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.1.0
nsh> /dev/ttyACM0: launching mavlink (HEARTBEAT v2 from SYSID:1 COMPID:200)
INFO  [mavlink] mode: Config, data rate: 100000 B/s on /dev/ttyACM0 @ 2000000B

nsh>
nsh> ERROR [uavcan] couldn't start parameter count: -2

nsh> commander arm
nsh> WARN  [PreFlightCheck] Arming denied! Flight termination active

nsh> 
```

julianoes, 2022-05-20:
It looks like this is related to https://github.com/PX4/PX4-Autopilot/pull/19419.

The problem is that the PR assumed lockdown means termination which is not the case for HITL where the outputs are "locked down" but nothing is in failsafe/termination.

I suggest to fix that for now as suggested in https://github.com/PX4/PX4-Autopilot/pull/19681.

julianoes, 2022-05-20:
Now the next problem I run into is that it refuses to take off:
```
INFO  [commander] Armed by external command
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] /fs/microsd/log/2022-05-20/04_40_59.ulg 
INFO  [logger] Opened full log file: /fs/microsd/log/2022-05-20/04_40_59.ulg
WARN  [mc_pos_control] invalid setpoints
WARN  [mc_pos_control] Failsafe: stop and wait
INFO  [commander] Takeoff detected
ERROR [commander] Remaining flight time low, returning to land  
INFO  [navigator] RTL HOME activated
INFO  [navigator] RTL: landing at home position.
INFO  [navigator] RTL: climb to 489 m (1 m above destination)   
INFO  [navigator] RTL: return at 489 m (1 m above destination)  
INFO  [navigator] RTL: land at destination
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing   
INFO  [logger] closed logfile, bytes written: 267737
```

julianoes, 2022-05-20:
So far I found two separate issues:

1. The flight time is estimated as too short. Presumably this happens because the battery percentage jumps from 100% to 70%, that's my current guess

2. The outputs are not correctly routed through to the simulator. I'm a bit confused what happened to the actuator output topics. The `pwm_out_sim` module seems to publish the `actuator_outputs_sim` topic, however, mavlink sends the normal `actuator_output` topic:

https://github.com/PX4/PX4-Autopilot/blob/c5bbf4553b1ca880b105a54af9a06f981d8a4ab9/src/drivers/pwm_out_sim/PWMSim.hpp#L93
https://github.com/PX4/PX4-Autopilot/blob/c5bbf4553b1ca880b105a54af9a06f981d8a4ab9/src/modules/mavlink/streams/HIL_ACTUATOR_CONTROLS.hpp#L69

I assume this has somewhat changed with the introduction of control allocation? Do you have any hints @bkueng or @dagar what's missing?

NicolasM0, 2022-05-20:
@julianoes I have seen the same issues

>     1. The flight time is estimated as too short. Presumably this happens because the battery percentage jumps from 100% to 70%, that's my current guess
--> the HITL sensor message generate a remaining time = 0. I have fixed that in local and I will make the pr.

>     2. The outputs are not correctly routed through to the simulator. I'm a bit confused what happened to the actuator output topics. The `pwm_out_sim` module seems to publish the `actuator_outputs_sim` topic, however, mavlink sends the normal `actuator_output` topic:

--> I had the same issue in SIH too (see [here](https://github.com/PX4/PX4-Autopilot/issues/18668)) and setting UAVCAN_ENABLE to 0 fix the issue for both SIH and HITL. But I think this is not a good fix. Maybe you have a better proposition?

NicolasM0, 2022-05-20:
@julianoes issue 1 should be fixed here: https://github.com/PX4/PX4-Autopilot/pull/19685

julianoes, 2022-05-20:
Thanks a lot @NicolasM0 for fixing 1.

closed 2022-05-21 by dagar

bkueng, 2022-05-23:
> I had the same issue in SIH too (see https://github.com/PX4/PX4-Autopilot/issues/18668) and setting UAVCAN_ENABLE to 0 fix the issue for both SIH and HITL. But I think this is not a good fix. Maybe you have a better proposition?

This won't be necessary anymore with dynamic control allocation.