PX4/PX4-Autopilot #19668: gazebo plane with SYS_CTRL_ALLOC enable errors with poll timeout

hamishwillee, 2022-05-18:
Modified SYS_CTRL_ALLOC to enabled control allocation (in source of param) and then tried the build either of these:
```
make px4_sitl gazebo_plane_catapult
make px4_sitl gazebo_plane
```
The output was as below, and QGC cannot connect.
```
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> ERROR [simulator] poll timeout 0, 22
poll timeout
poll timeout
ERROR [simulator] poll timeout 0, 22
poll timeout
ERROR [simulator] poll timeout 0, 22
...
```
Note that if SYS_CTRL_ALLOC  is not enabled the frame works. Also other frames like VTOL and iris work with this setting.

FYI @bkueng

hamishwillee, 2022-05-18:
PS @bkueng  . BTW, this is supposed to be permanently enabled in PX4 v1.13. IF I build that tag will I get control allocation "as it should be" for an end user?

Note that enabling SYS_CTRL_ALLOC does appear to allow the following cases 
```
make px4_sitl gazebo_tailsitter
make px4_sitl gazebo_standard_vtol
make px4_sitl gazebo_tailsitter
make px4_sitl gazebo_tiltrotor
make px4_sitl gazebo_rover
make px4_sitl gazebo_r1_rover
```

Simulator worked, but not sure if the setup is correct for `gazebo_believer` - just has one motor.

bkueng, 2022-05-18:
I will have a look.

Jaeyoung-Lim, 2022-05-27:
I can confirm that this happens on my setup too

closed 2022-05-31 by bkueng