PX4/PX4-Autopilot #19671: EKF2 multiple height source fusion proposal

bresch, 2022-05-18:
There are several possible ways to incorporate height data from different sources into a navigation filter. Currently, a primary source (baro, GNSS, range) is defined by a parameter `EKF2_HGT_MODE` and the other sources are used as fallback. A special behavior is also possible with `EKF2_RNG_AID` where the estimator switches to "range primary" when the vehicle isn't moving horizontally using a constant terrain height assumption. Recently we added a baro bias estimator to mitigate the baro drift using GNSS data when in "baro primary" mode.

Switching between different primary height sources requires to accumulate an offset for each source and the estimate can freely drift away from an absolute reference when the switch occurs too frequently (ratcheting effect observed when using range aid). Also, all sensors have different error characteristics, meaning that we can reduce the error of each sensor if we combine them together (e.g.: estimate the baro bias using GNSS data or range finder when not moving, estimate the terrain height using GNSS and baro data when moving, ...).

First, let's define the different measurements and variables used in the height fusion process and how they are currently used:
- A startup, the EKF origin is initialized at 0 and referenced using the current barometric height (`baro_off`).
- When the GNSS passes the quality checks, the height difference between the EKF origin and the GNSS origin is set (`GNSS alt ref`)
- The baro bias is estimated using the GNSS measurement given that `baro_off` and `GNSS alt ref` are fixed
- The ground position is estimated using the current altitude and the range measurement (it can also be estimated using optical flow but this isn't covered here)

[image]

My proposal would be to not only combine the measurements to estimate the bias between them, but also fuse all the possible measurements in the filter, without creating a loop. The general idea is to always fuse GNSS and baro altitudes when selected and to use the range measurement as a source of aiding for the baro bias estimator when a constant terrain height assumption can be done (i.e.: no lateral motion).

| Mode                  | GNSS | Barometer | Range finder |   | Estimated                                                                                                     | Fixed variables                              | Fused sources into EKF2 |
|-----------------------|:----:|:---------:|:------------:|---|---------------------------------------------------------------------------------------------------------------|----------------------------------------------|-------------------------|
| Outdoor  standard    |   X  |     X     |       -      |   | baro bias                                                                                                     |                                              | GNSS, baro              |
| Outdoor w/ rng aiding |   X  |     X     |       X      |   | -motion: baro bias (from GNSS), terrain (from terrain est)<br>-static: baro bias (from GNSS and range finder) | -motion: -<br>-static: terrain | GNSS, baro              |
| Outdoor w/o baro      |   X  |     -     |       X      |   | terrain                                                                                                       |                                              | GNSS                    |
| Indoor                |   -  |     X     |       X      |   | -motion: terrain<br>-static: baro bias                                                                        | -motion: baro bias<br>-static: terrain       | baro                    |

Notes:
- the other trivial modes "GNSS only", "baro only" and "range only" are also possible but not represented in the table
- in "Outdoor standard" mode, available range finder data would still be used to estimate the terrain height but will not affect the EKF height estimate at all
- "Outdoor w/o baro" is the same as "GNSS only" since the range data can only be used to estimate the terrrain
- in "Outdoor w/ rng aiding" it might even be better to run the terrain estimator even when not moving to prevent the GNSS and range data from fighting "for ever" on the baro bias (the constant innovation residual would then leak into the terrain estimate).
- the same measurement is potentially used at several places so we need to update the states in "parallel" (e.g.: first fuse the baro compensated by its bias estimate and the GNSS before updating the bias estimate)

ecmnet, 2022-05-21:
I like that idea. Maybe Vision height should be considered as well. My use case is outdoors w/o GNSS. Due to light conditions, vision height is not always available, a fallback to range and/or baro would be nice in such cases, as in narrow situations a reliable altitude is vital. In previous version of EKF2, I have provided a solution to this (which still works fine for me).

dagar, 2022-05-21:
> In previous version of EKF2, I have provided a solution to this (which still works fine for me).

@ecmnet Can you link to that here?

ecmnet, 2022-05-22:
Fallback of EV height: https://github.com/PX4/PX4-ECL/pull/994

bresch, 2022-05-23:
> Maybe Vision height should be considered as well

In the proposal I made above, vision can be considered as "range finder" since they both measure the height of the drone above terrain. If the baro+vision height is active, then the vision is correcting for the baro bias when static and the baro itself is fused in the EKF (see table above). If only vision is selected, you'll still have a fallback to another available height source (e.g.: baro) in case the vision data isn't available anymore.

ecmnet, 2022-05-24:
Two comments from my side:
1. I doubt that the assumption is true, that vision is a source of height above terrain. At least in my case it returns height information independent on terrain altitude. Example showing terrain altitude, vision z and rangefinder

[image]

2. Fusing baro only in "indoor" mode would add baro noise even if the bias is compensated. I doubt that using this would allow cm precision in altitude controlled modes, especially indoors.

Add least a mode 'external' (similar to range only) should be added, which allows the range finder as source for terrain only and to control altitude externally. This would allow to implement fallbacks/fusion on companion side.

ecmnet, 2022-05-24:
Some  comments from my side:

1. I think the assumption that vision provides terrain altitude is not correct. It is the advantage of vision controlled height compared to the range finder, that vision is not terrain dependent. Adding a rangefinder to vision would allow to estimate terrain. If vision fails, a fallback to RF would compensate this until vision is re-initialized; terrain would be assumed as constant (which is of course not always true, but would be the best guess). This fallback  might not be required, if a fallback to baro (with bias compensation) provides sufficient accurracy and does not lead to altitude jumps.
2. Fusing baro only in "indoor" mode would not allow cm altitude accuracy as baro noise would be fused, even if the bias would be compensated. Controlling height in a simple "range finder" mode however would not have any fallback in case the height source fails.

ecmnet, 2022-06-06:
An additional note on Baro with RF aid enabled:
On ground and disarmed, the EKF2 mode is baro, leading to a moving LPOS-Z. Why is that? With RF aid enabled, relative altitude should be 0, LPOS-Z stable. Especially to initialise LPOS origin, that should be ensured.

closed 2022-10-26 by bresch