PX4/PX4-Autopilot #19678: roslaunch mavros_posix_sitl.launch WARN [PreFlightCheck] Preflight Fail: Accel 1 inconsistent - Check Cal

erha527, 2022-05-19:
Hi~
I am having issues running the mavros_posix_sitl.launch on my px4-mavros-gazebo sitl. When I roslaunch mavros_posix_sitl.launch ,I meet an issue that __

WARN [PreFlightCheck] Preflight Fail: Accel 1 inconsistent - Check Cal

so that I can not arm the vehicle ,and I can also not use the qgc to arm.But when I change the

_<arg name="vehicle" default="iris"/> to <arg name="vehicle" default="iris_rplidar"/>_in the mavros_posix_sitl.launch , it makes useful.
I did not edit the iris.sdf, and I need to use iris.sdf model.Could you help me?

Jaeyoung-Lim, 2022-05-27:
> _ to _in the mavros_posix_sitl.launch , it makes useful.

Could you clarify what this is referring to?

AlexWUrobot, 2023-02-07:
It may be helpful to clean the `~/.ros`
ref: https://github.com/PX4/PX4-Autopilot/issues/18299

simojo, 2024-02-01:
> It may be helpful to clean the `~/.ros`

@AlexWUrobot would this be done using `rosclean` or by some other method? (It's not completely clear which files are safe to clean)