PX4/PX4-Autopilot #19682: Failsafe Enabled: No Local Position my drone crashed sideways

airsimdevdd, 2022-05-20:
my drone crashed sideways after showing failsafe enabled no local position here is my attached photos

mcsauder, 2022-05-20:
Hi @airsimdevdd , could you post a link to the full log here?

airsimdevdd, 2022-06-13:
for some reasons i can't  post full log here

bresch, 2022-06-13:
All the plots you share suggest an issue with the magnetic field measured by the magnetometer: The norm of the magnetic field seems correlated with the thrust and its norm is way too large (almost 10Gs while the earth mag field is ~0.4-0.6Gs). I guess you already suspected the mag since you shared those plots.
You probably need to move the magnetometer further away from cables, batteries and other components that can radiate a large magnetic field.

bresch, 2022-06-13:
Since it isn't a software issue, I would suggest to continue on the forum and in PX4 slack.

closed 2022-06-13 by bresch