PX4/PX4-Autopilot #19684: vehicle_command working on simulation but not with real drone

MrPrietix, 2022-05-20:
Pixhawk v5
  PX4 version: 1.13.0 alpha

Companion computer
  Ubuntu: 20.04
  Ros2: foxy

I have been changing the home position and arming the drone in simulation with the following code:
[offboard_control.txt](https://github.com/PX4/PX4-Autopilot/files/8737721/offboard_control.txt)

In simulation it works without no bugs. But when I perform the same action with a real drone it does not arm nor change the home position.

To evaluate the code on the drone first I launch the micrortps client and agent, I make sure that I receive some ros2 topics on the companion computer to check the micortps bridge. Consequently, I execute the code but the drone is not arming nor chaning the home position. 

Note: I do the test with the real drone outside to have a home position, then once I have a home position, I execute the code to see if it works.

Could it be because of the PX4 version?

Thanks,

MrPrietix

sagardalai, 2022-07-04:
Hi @MrPrietix 

Can you help me to establish the communication between PX4 in Pixhawk and the companion computer(ROS2)? I have tried different ways but failed. 
1) https://discuss.px4.io/t/ros2-offboard-on-hardware/26338/2

Thanks regard

sagardalai, 2022-07-10:
Hay @MrPrietix 
As mentioned above I have tried to follow the steps given in this link i.e. https://discuss.px4.io/t/ros2-offboard-on-hardware/26338/2

I have FMU V5 controller on a hexacopter trying to communicate with the companion computer but failed.

Please give me an idea whether am I missing something or am I in wrong track.

MrPrietix, 2022-07-10:
Hei Sagar

With this command (micrortps_client start -t UART) on the discuss they are
not specifying the port, and it is not a permanent setup, so every time you
power up the pixhawk it is going to lose that configuration. So the way to
do it is to go into the SD card of the flight controller and create a
folder /etc on the root dir, then in /etc create a file with the name
extras.txt. Consequently, paste the following in the extras.txt. Here you
can decide which port you want, and comment on the one that you do not use.
I think this is the only thing missing. Let me know if you have some more
problems. By the way, how did you find me?

Cheers,

Sergio Prieto

set +e
# For UART communication over the TELEM port:
micrortps_client start -d /dev/ttyS4 -b 921600

# For UART communication over the TELM2 port:
micrortps_client start -d /dev/ttyS2 -b 921600
set -e

sagardalai, 2022-07-15:
Hi @MrPrietix 
Hope that you are well and sorry for the late response. These are the steps that I followed for the setup. Please check the steps and let me know if I am missing anything or any other suggestions.

Problem: Unable to establish communication between the flight controller and companion computer.

companion computer OS : focal
Flight controller : FMU v5+
ROS2 : Foxy

1) Install ROS2 foxy : (https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html)
2) foonathan_memory_vendor:
3) Fast-RTPS-Gen : (https://docs.px4.io/v1.12/en/dev_setup/fast-dds-installation.html#fast-rtps-gen)
4) px4_ros_com_ros2 :  https://docs.px4.io/v1.12/en/ros/ros2_comm.html
5) PX4-Autopilot :

Final Steps : https://discuss.px4.io/t/building-px4-ros2-bridge-on-hardware-pixhawk-4/24266/2

[image]

unable to establish communication.

Please check the above steps.
How can I contact you apart from this issue?