PX4/PX4-Autopilot #19723: make posix_stil_default gazebo error

f123h, 2022-05-26:
FAILED: cd /home/salad/Firmware/build/posix_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -GNinja /home/salad/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch
/home/salad/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure

-- ==> add_subdirectory(.)
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  add_subdirectory not given a binary directory but the given source
  directory "/home/salad/Firmware/Tools/sitl_gazebo" is not a subdirectory of
  "/home/salad/Firmware/Tools/sitl_gazebo".  When specifying an out-of-tree
  source a binary directory must be explicitly specified.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/salad/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/salad/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

Jaeyoung-Lim, 2022-05-26:
@f123h If you are trying to run SITL, the correct command is as written in the instructions: https://docs.px4.io/master/en/simulation/gazebo.html
```
make px4_sitl gazebo
```

f123h, 2022-05-27:
> @f123h If you are trying to run SITL, the correct command is as written in the instructions: https://docs.px4.io/master/en/simulation/gazebo.html
> 
> ```
> make px4_sitl gazebo
> ```

Oh,thanks your help. I tried the command but other errors have occurred

make px4_sitl gazebo
Makefile:378: *** "px4_sitl cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets."。 停止。

Jaeyoung-Lim, 2022-05-27:
@f123h Have you tried what the command log suggests? (`make help` or make list_config_targets)

are you sure you are using the latest PX4 version?

f123h, 2022-05-28:
> @f123h Have you tried what the command log suggests? (`make help` or make list_config_targets)
> 
> are you sure you are executing the command at the root of this repository?

>I am sure.
>I have tried the command(make list_config_targets)
salad@123:~/Firmware$ make list_config_targets
[nuttx_]aerocore2_default
[nuttx_]aerofc-v1_default
[nuttx_]aerofc-v1_rtps
[nuttx_]auav-x21_default
[nuttx_]crazyflie_default
[nuttx_]esc35-v1_default
[nuttx_]mindpx-v2_default
[nuttx_]nxphlite-v3_default
[nuttx_]omnibus-f4sd_default
[nuttx_]px4cannode-v1_default
[nuttx_]px4esc-v1_default
[nuttx_]px4fmu-v2_default
[nuttx_]px4fmu-v2_lpe
[nuttx_]px4fmu-v2_test
[nuttx_]px4fmu-v3_default
[nuttx_]px4fmu-v3_rtps
[nuttx_]px4fmu-v4_default
[nuttx_]px4fmu-v4pro_default
[nuttx_]px4fmu-v4pro_rtps
[nuttx_]px4fmu-v4_rtps
[nuttx_]px4fmu-v5_default
[nuttx_]px4fmu-v5_rtps
[nuttx_]px4io-v2_default
[nuttx_]px4nucleoF767ZI-v1_default
[nuttx_]px4-same70xplained-v1_default
[nuttx_]px4-stm32f4discovery_default
[nuttx_]s2740vc-v1_default
[nuttx_]tap-v1_default
posix_bebop_default
posix_eagle_default
posix_eagle_hil
posix_eagle_legacy
posix_eagle_muorb
posix_excelsior_default
posix_excelsior_legacy
posix_ocpoc_cross
posix_ocpoc_ubuntu
posix_rpi_cross
posix_rpi_cross_no_shield
posix_rpi_native
posix_sitl_broadcast
posix_sitl_default
posix_sitl_ekf2
posix_sitl_inav
posix_sitl_lpe
posix_sitl_replay
posix_sitl_rtps
posix_sitl_shell
posix_sitl_test
qurt_eagle_default
qurt_eagle_hello
qurt_eagle_hil
qurt_eagle_legacy
qurt_eagle_muorb
qurt_eagle_test
qurt_eagle_travis
qurt_excelsior_default
qurt_excelsior_legacy

Jaeyoung-Lim, 2022-05-28:
@f123h Which version of PX4 do you have? This looks different from the current master

```
$ make list_config_targets
airmind_mindpx-v2[_default]
ark_can-flow_canbootloader
ark_can-flow[_default]
ark_can-gps_canbootloader
ark_can-gps[_default]
ark_cannode_canbootloader
ark_cannode[_default]
ark_can-rtk-gps_canbootloader
ark_can-rtk-gps_debug
ark_can-rtk-gps[_default]
atl_mantis-edu[_default]
av_x-v1[_default]
beaglebone_blue[_default]
bitcraze_crazyflie21[_default]
bitcraze_crazyflie[_default]
cuav_can-gps-v1_canbootloader
cuav_can-gps-v1[_default]
cuav_nora_bootloader
cuav_nora[_default]
cuav_x7pro_bootloader
cuav_x7pro[_default]
cuav_x7pro_test
cubepilot_cubeorange_bootloader
cubepilot_cubeorange[_default]
cubepilot_cubeorange_test
cubepilot_cubeyellow[_default]
cubepilot_io-v2[_default]
diatone_mamba-f405-mk2[_default]
emlid_navio2[_default]
freefly_can-rtk-gps_canbootloader
freefly_can-rtk-gps[_default]
holybro_can-gps-v1_canbootloader
holybro_can-gps-v1[_default]
holybro_durandal-v1_bootloader
holybro_durandal-v1[_default]
holybro_kakutef7[_default]
holybro_kakuteh7_bootloader
holybro_kakuteh7[_default]
holybro_pix32v5[_default]
matek_gnss-m9n-f4_canbootloader
matek_gnss-m9n-f4[_default]
matek_h743_bootloader
matek_h743[_default]
matek_h743-mini_bootloader
matek_h743-mini[_default]
matek_h743-slim_bootloader
matek_h743-slim[_default]
modalai_fc-v1[_default]
modalai_fc-v1_rtps
modalai_fc-v2_bootloader
modalai_fc-v2[_default]
mro_ctrl-zero-classic_bootloader
mro_ctrl-zero-classic[_default]
mro_ctrl-zero-f7[_default]
mro_ctrl-zero-f7-oem[_default]
mro_ctrl-zero-h7_bootloader
mro_ctrl-zero-h7[_default]
mro_ctrl-zero-h7-oem_bootloader
mro_ctrl-zero-h7-oem[_default]
mro_ctrl-zero-h7-oem_rtps
mro_ctrl-zero-h7_rtps
mro_pixracerpro_bootloader
mro_pixracerpro[_default]
mro_pixracerpro_rtps
mro_x21-777[_default]
mro_x21[_default]
nxp_fmuk66-e[_default]
nxp_fmuk66-e_rtps
nxp_fmuk66-e_socketcan
nxp_fmuk66-v3[_default]
nxp_fmuk66-v3_rtps
nxp_fmuk66-v3_socketcan
nxp_fmuk66-v3_test
nxp_fmurt1062-v1[_default]
nxp_ucans32k146_canbootloader
nxp_ucans32k146[_default]
nxp_ucans32k146_servo
omnibus_f4sd[_default]
omnibus_f4sd_icm20608g
px4_fmu-v2[_default]
px4_fmu-v2_fixedwing
px4_fmu-v2_lto
px4_fmu-v2_multicopter
px4_fmu-v2_rover
px4_fmu-v3[_default]
px4_fmu-v3_test
px4_fmu-v4[_default]
px4_fmu-v4pro[_default]
px4_fmu-v4pro_test
px4_fmu-v4_test
px4_fmu-v5_cryptotest
px4_fmu-v5_cyphal
px4_fmu-v5_debug
px4_fmu-v5[_default]
px4_fmu-v5_lto
px4_fmu-v5_protected
px4_fmu-v5_rtps
px4_fmu-v5_stackcheck
px4_fmu-v5_test
px4_fmu-v5_uavcanv0periph
px4_fmu-v5x[_default]
px4_fmu-v5x_rtps
px4_fmu-v5x_test
px4_fmu-v6c_bootloader
px4_fmu-v6c[_default]
px4_fmu-v6u_bootloader
px4_fmu-v6u[_default]
px4_fmu-v6x_bootloader
px4_fmu-v6x[_default]
px4_io-v2[_default]
px4_raspberrypi[_default]
px4_sitl[_default]
px4_sitl_nolockstep
px4_sitl_replay
px4_sitl_rtps
px4_sitl_test
raspberrypi_pico[_default]
scumaker_pilotpi_arm64
scumaker_pilotpi[_default]
sky-drones_smartap-airlink[_default]
spracing_h7extreme[_default]
uvify_core[_default]
```

f123h, 2022-05-29:
> @f123h Which version of PX4 do you have? This looks different from the current master

The version of PX4 is 1.8.0,  ubuntu 16.04,  gazebo 7.16.1 ,  ros (kinetic)
$ rosparam get /rosdistro
'kinetic
  '
--------------------------------------------------------------------------------
$ gazebo -v
Gazebo multi-robot simulator, version 7.16.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.16.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
----------------------------------------------------------------------
$ make posix_sitl_default gazebo
-- PX4 VERSION: v1.8.0
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-553-g1a11068
-- Configuring done
-- Generating done
-- Build files have been written to: /home/salad/Firmware/build/posix_sitl_default
ninja: Entering directory `/home/salad/Firmware/build/posix_sitl_default'
[328/645] Performing configure step for 'sitl_gazebo'
FAILED: cd /home/salad/Firmware/build/posix_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -GNinja /home/salad/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/salad/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/salad/Firmware/build/posix_sitl_default/build_gazebo/devel
-- Using CMAKE_PREFIX_PATH: /home/salad/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/salad/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/salad/Firmware/build/posix_sitl_default/build_gazebo/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - mavlink_sitl_gazebo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'mavlink_sitl_gazebo'
-- ==> add_subdirectory(.)
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  add_subdirectory not given a binary directory but the given source
  directory "/home/salad/Firmware/Tools/sitl_gazebo" is not a subdirectory of
  "/home/salad/Firmware/Tools/sitl_gazebo".  When specifying an out-of-tree
  source a binary directory must be explicitly specified.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/salad/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/salad/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[328/645] Building CXX object src/modu...uORB_tests.dir/uORBTest_UnitTest.cpp.o
ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

f123h, 2022-05-29:
I use the following command (apt-cache search ros-kinetic) and I find the version of gazebo is different from previous the version (gazebo -v)

$ apt-cache search ros-kinetic
ros-kinetic-gazebo8-dev - Provides a cmake config for the default version of Gazebo for the ROS distribution.
ros-kinetic-gazebo8-msgs - Message and service data structures for interacting with Gazebo from ROS.
ros-kinetic-gazebo8-plugins - Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.
ros-kinetic-gazebo8-ros - Provides ROS plugins that offer message and service publishers for interfacing with Gazebo through ROS.
ros-kinetic-gazebo8-ros-control - gazebo_ros_control
ros-kinetic-gazebo8-ros-pkgs - Interface for using ROS with the Gazebo simulator.
ros-kinetic-gazebo9-dev - Provides a cmake config for the default version of Gazebo for the ROS distribution.
ros-kinetic-gazebo9-msgs - Message and service data structures for interacting with Gazebo from ROS.
ros-kinetic-gazebo9-plugins - Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.
ros-kinetic-gazebo9-ros - Provides ROS plugins that offer message and service publishers for interfacing with Gazebo through ROS.
ros-kinetic-gazebo9-ros-control - gazebo_ros_control
ros-kinetic-gazebo9-ros-pkgs - Interface for using ROS with the Gazebo simulator.
----------------------------------------------------------------------------------------------------
$ gazebo -v
Gazebo multi-robot simulator, version 7.16.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.16.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Jaeyoung-Lim, 2022-05-29:
@f123h What is the version you have of this repository?

f123h, 2022-05-29:
> @f123h You have shared versions of other dependencies that PX4 uses. However, not the version of PX4. What is the version you have of this repository?

I am not sure the version.
/**
 * Convert a version tag string to a number
 * @param tag version tag in one of the following forms:
 *            - vendor: v1.4.0-0.2.0
 *            - dev: v1.4.0rc3-7-g7e282f57
 *            - rc: v1.4.0rc4
 *            - release: v1.4.0
 *            - linux: 7.9.3
 * @return version in the form 0xAABBCCTT (AA: Major, BB: Minor, CC: Patch, TT Type @see FIRMWARE_TYPE)
 */
__EXPORT uint32_t version_tag_to_number(const char *tag);

/**
 * get the PX4 Firmware version
 * @return version in the form 0xAABBCCTT (AA: Major, BB: Minor, CC: Patch, TT Type @see FIRMWARE_TYPE)
 */
__EXPORT uint32_t px4_firmware_version(void);

/**
 * Convert a version tag string to a vendor version number
 * @param tag version tag in one of the following forms:
 *            - vendor: v1.4.0-0.2.0
 *            - dev: v1.4.0rc3-7-g7e282f57
 *            - rc: v1.4.0rc4
 *            - release: v1.4.0
 *            - linux: 7.9.3
 * @return version in the form 0xAABBCCTT (AA: Major, BB: Minor, CC: Patch, TT Type @see FIRMWARE_TYPE)
 */

Jaeyoung-Lim, 2022-05-29:
@f123h Tou can share which git commit you are on: 
https://git-scm.com/book/en/v2/Git-Basics-Viewing-the-Commit-History

f123h, 2022-05-29:
> Tou can share which git commit you are on:

$ git log
commit 96443b3cf3e3adaec8b2ebb6c83a19a92d40a13f
Author: Lorenz Meier <lorenz@px4.io>
Date:   Tue Jun 19 08:23:34 2018 +0200

    Update README

commit d1343b0ccbc2ab1f676f36868250cb4e7b27f277
Author: Beat Küng <beat-kueng@gmx.net>
Date:   Mon Jun 18 11:40:55 2018 +0200

    calibration_routines: fix 'Command denied during calibration' error message
    
    The uorb subscribe logic got changed for queued topics with
    https://github.com/PX4/Firmware/pull/9436, meaning an orb_subscribe will
    return past messages as well now.
    
    This lead to an error 'Command denied during calibration' for the previously
    received calibration start command.

commit 1fac3a1cee914f22f4bfefac3976483af0056040
Author: PX4 Jenkins <bot@pixhawk.org>
Date:   Sat Jun 16 21:27:20 2018 -0400

    Update submodule sitl_gazebo to latest Sat Jun 16 21:27:20 EDT 2018
    
        - sitl_gazebo in PX4/Firmware (ebc40067c71e158646cc0fece422b0432c18878b): https://github.com/PX4/sitl_gazebo/commit/8a3166bf5d315274cfd503a75880e9837faa9694
        - sitl_gazebo current upstream: https://github.com/PX4/sitl_gazebo/commit/b5a92095bfb1b6a218a80f58ad9fe09c89ef4598
        - Changes: https://github.com/PX4/sitl_gazebo/compare/8a3166bf5d315274cfd503a75880e9837faa9694...b5a92095bfb1b6a218a80f58ad9fe09c89ef4598
    
        b5a9209 2018-06-10 TSC21 - travis_ci: Add OSX build

commit ebc40067c71e158646cc0fece422b0432c18878b
:
    https://github.com/PX4/Firmware/pull/9436, meaning an orb_subscribe will
    return past messages as well now.
    
    This lead to an error 'Command denied during calibration' for the previously
    received calibration start command.

Jaeyoung-Lim, 2022-05-29:
@f123h exactly, you are using a version roughly 4 years old. Have you tried the latest master? Recent versions have dropped support for Ubuntu 16.04 so you might need to update your operating system

f123h, 2022-05-29:
> @f123h exactly, you are using a version roughly 4 years old. Have you tried the latest master? Recent versions have dropped support for Ubuntu 16.04 so you might need to update your operating system

But  the command (make posix_sitl_default jmavsim) is successful in ubuntu(16.04). Now, I don't  know what problems are.