PX4/PX4-Autopilot #19731: Need to update libuavcan depedency to fix build with Python 3.10

dlech, 2022-05-27:
## Describe the bug

I've just tried building the firmware on Ubuntu 22.04 which has Python 3.10. It results in the following error:

```
FAILED: libuavcan_dsdlc_run.stamp /home/david/work/PX4-Autopilot/build/px4_fmu-v3_default/libuavcan_dsdlc_run.stamp 
cd /home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan && /home/david/work/PX4-Autopilot/.venv/bin/python3 /home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc test/dsdl_test/root_ns_a test/dsdl_test/root_ns_b /home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/../dsdl/uavcan -Oinclude/dsdlc_generated && /usr/bin/cmake -E touch /home/david/work/PX4-Autopilot/build/px4_fmu-v3_default/libuavcan_dsdlc_run.stamp
Traceback (most recent call last):
  File "/home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
  File "/home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/__init__.py", line 15, in <module>
    import uavcan.transport as transport
  File "/home/david/work/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/transport.py", line 203, in <module>
    class ArrayValue(BaseValue, collections.MutableSequence):
AttributeError: module 'collections' has no attribute 'MutableSequence'
```

The workaround is to make the following change:

```diff
diff --git a/uavcan/transport.py b/uavcan/transport.py
index 1c914de..9cc0d42 100644
--- a/uavcan/transport.py
+++ b/uavcan/transport.py
@@ -200,7 +200,7 @@ class PrimitiveValue(BaseValue):
             self._bits = format(int_value, "0" + str(self.type.bitlen) + "b")
 
 
-class ArrayValue(BaseValue, collections.MutableSequence):
+class ArrayValue(BaseValue, collections.abc.MutableSequence):
     def __init__(self, uavcan_type, tao=False, *args, **kwargs):
         super(ArrayValue, self).__init__(uavcan_type, *args, **kwargs)
         value_bitlen = getattr(self.type.value_type, "bitlen", None)
```

Some digging shows that we are using a 7-year old version of https://github.com/dronecan/pydronecan which is a submodule of https://github.com/dronecan/libuavcan. The issue has been fixed upstream long ago, so it is just a matter of updating the depedency.

dagar, 2022-05-28:
Did you try simply bumping the pydronecan submodule or did you hit other issues? 

I'd like to get all of these libuavcan dependencies updated to latest, but last time I looked we had a few small issues to work through.

dlech, 2022-05-28:
No I didn't try it yet.

dlech, 2022-05-29:
Related issue: https://github.com/dronecan/libuavcan/issues/3

PR #20156 (2022-09-10): libuavcan: update submodule
This fixes a Python 3.10 issue for me.

Fixes #19731.

I don't know what this update entails, I have done no testing except compiling `make px4_fmu-v5`.
closed 2022-09-10 by dagar