PX4/PX4-Autopilot #19733: No RC Input on FMU-6C

ryanjAA, 2022-05-27:
Running Master from today, there is no RC input on fmu-6c. Tried an X8R receiver but there is no rc_input module. Is this by design and there is something else or omitted and something that simply needs to be added in.

dagar, 2022-05-28:
Was it ever working? Can you run `px4io status`?

ryanjAA, 2022-05-28:
Not running. See below:

NuttShell (NSH) NuttX-10.2.0
nsh> px4io status
INFO  [px4io] not running
nsh> ver all
HW arch: PX4_FMU_V6C
HW type: V6C00
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: 3bafaa451dc7f6f482acde034ede669118434c9c
FW version: 1.13.0 80 (17629312)
OS: NuttX
OS version: Release 10.2.0 (167903487)
OS git-hash: c5c7d2b4f26f52f1dfb425ebde83328606b65d4f
Build datetime: May 27 2022 15:54:40
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0006000000003633373332305106001b002d
MCU: STM32H7[4|5]xxx, rev. V
nsh>

ryanjAA, 2022-05-29:
@vincentpoont2

davids5, 2022-05-29:
@dagar @ryanjAA @vincentpoont2 

This was tested today. 

```

ryanjAA, 2022-05-29:
Will update the io then post output. Although hearing this, it’s probably fine since this hardware just arrived (and is likely a new revision).

davids5, 2022-05-29:
What is the SYS_AUTOSTART set to?

vincentpoont2, 2022-05-29:
@ryanjAA Did you have a airframe selected when you use `px4io status` ?

ryanjAA, 2022-05-30:
io update did it

NuttShell (NSH) NuttX-10.2.0
nsh> px4io status
protocol 5 hardware 2 bootloader 3 buffer 64B crc 0x6210128859
8 controls 8 actuators 18 R/C inputs 2 analog inputs
 px4io_status
    timestamp: 94218281 (0.594571 seconds ago)
    voltage_v: 0.0000
    rssi_v: 1.1290
    free_memory_bytes: 1464
    pwm: [900, 900, 900, 900, 0, 0, 0, 0]
    pwm_disarmed: [900, 900, 900, 900, 0, 0, 0, 0]
    pwm_failsafe: [0, 0, 0, 0, 0, 0, 0, 0]
    pwm_rate_hz: [400, 400, 400, 400, 50, 50, 50, 50]
    raw_inputs: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    status_arm_sync: True
    status_failsafe: False
    status_fmu_initialized: True
    status_fmu_ok: True
    status_init_ok: True
    status_outputs_armed: True
    status_raw_pwm: True
    status_rc_ok: False
    status_rc_dsm: False
    status_rc_ppm: False
    status_rc_sbus: False
    status_rc_st24: False
    status_rc_sumd: False
    status_safety_off: True
    alarm_pwm_error: False
    alarm_rc_lost: True
    arming_failsafe_custom: True
    arming_fmu_armed: False
    arming_fmu_prearmed: False
    arming_force_failsafe: False
    arming_io_arm_ok: True
    arming_lockdown: False
    arming_termination_failsafe: False

0 raw R/C inputs
ADC inputs 0 1381
features 0x0004 RSSI_ADC
sbus rate 72
debuglevel 0

INFO  [mixer_module] Param prefix: PWM_MAIN
control latency: 34888 events, 13066671us elapsed, 374.53us avg, min 326us max 970us 13.627us rms
INFO  [mixer_module] Switched to rate_ctrl work queue
INFO  [mixer_module] Mixer loaded: yes
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 900, failsafe: 0, disarmed: 900, min: 1075, max: 1950
Channel 1: value: 900, failsafe: 0, disarmed: 900, min: 1075, max: 1950
Channel 2: value: 900, failsafe: 0, disarmed: 900, min: 1075, max: 1950
Channel 3: value: 900, failsafe: 0, disarmed: 900, min: 1075, max: 1950
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 1000, max: 2000
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 1000, max: 2000
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 1000, max: 2000
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 1000, max: 2000
nsh>

closed 2022-05-30 by ryanjAA