PX4/PX4-Autopilot #19740: Add sensor type support to Kconfig as dependency for including drivers, modules, subsets of sensors hub and estimators

dagar, 2022-05-29:
If we add a support for sensor types within kconfig we could then use it optionally strip down the system in many cases.

Airspeed for example
 - differential pressure sensor drivers
 - differential_pressure, airspeed, airspeed_validated .msg files (uORB topics)
 - airspeed_selector module
 - `sensors` hub airspeed support (and calibration parameters)
 - commander airspeed calibration code and checks
 - ekf2 airspeed fusion
 - etc

The overall flash savings would be worth it for small quadcopters or specialized use cases like CAN nodes.

@PetervdPerk-NXP @AlexKlimaj @bresch

dagar, 2022-05-29:
Somewhat related https://github.com/PX4/PX4-Autopilot/pull/19714.

PetervdPerk-NXP, 2022-05-30:
Kconfig options are already exposed to CMake and C/C++ (#19116).
So for commander you could use #ifdef to guard the airspeed calibration code behind that.