PX4/PX4-Autopilot #19747: Control Surfaceless VTOL Quad Tailsitters

aoberai, 2022-05-30:
VTOL Quad Tailsitters

## Problem

VTOL Quad Tailsitters without elevons (control surfaceless) are not supported by PX4. Examples of such aerial vehicles with this design include the Swan K1. A video of it can be seen here: https://www.youtube.com/watch?v=bRfDnXAc31A&amp;ab_channel=MohammadAdib&ab_channel=MohammadAdib

## Solution
Allow for pitch and roll differential thrust to be applied by quad tailsitters (which would allow for rotational control without elevons).

## Additional context
Ardupilot has this functionality. The Swan K1 actually uses a modified version of PX4, though that was not upstreamed.

Jaeyoung-Lim, 2022-10-17:
Also in line with this feature :smile:

aoberai, 2022-10-17:
@sfuhrer had done some work earlier to make this work for the SWAN K1 - forgot which branch that development was occurring on.

closed 2022-11-07 by Jaeyoung-Lim