PX4/PX4-Autopilot #19753: SYS_STATUS GPS enabled flag not set by default

julianoes, 2022-06-01:
PX4 does - by default - not set the flag SYS_STATUS.onboard_control_sensors_enabled for GPS.
https://mavlink.io/en/messages/common.html#SYS_STATUS

It only sets the flag if the param [COM_ARM_WO_GPS](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_ARM_WO_GPS) is set to 0 meaning "Require GPS to arm".

I can see that by setting this param to 0 we make sure to only fly with GPS, however, I would argue, that whenever you switch to a mode which requires GPS, and relies on it, you essentially have the GPS "enabled". Or you could go as far as that if you run the GPS drivers, then clearly the GPS is "enabled.

For context: I've stumbled on this when implementing this flag in MAVSDK: https://github.com/mavlink/MAVSDK/pull/1794
I can work around the oddity for now but - in my opinion - it is not really according to how I'd interpret the spec.

dagar, 2022-06-01:
I don't really know how to interpret it either, but one idea I had was to capture configuration parameters per GNSS instance.
https://github.com/PX4/PX4-Autopilot/pull/19402

junwoo091400, 2022-07-11:
Oops that would be about the `health`. Sorry!

junwoo091400, 2022-07-11:
Or with `SYS_HAS_GPS` parameter? As that's how it's used in the `ekf2Check`:

https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/commander/Arming/PreFlightCheck/checks/ekf2Check.cpp#L165-L166