PX4/PX4-Autopilot #19756: DShot Motor Reverse Direction uses wrong DShot command

BA-ED, 2022-06-01:
## Describe the bug
Commands `DShot_cmd_spin_direction_1` and `DShot_cmd_spin_direction_2` are meant to be used for setting the motor direction to be saved to eeprom.
Commands `DShot_cmd_spin_direction_normal` and `DShot_cmd_spin_direction_reversed` are meant to be used for setting the motor direction temporarily (e.g. for motor testing and turtle mode).
[Here](https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/dshot/DShot.cpp#L755) is the code for each command. As you can see, it uses `DShot_cmd_spin_direction_reversed` instead of `DShot_cmd_spin_direction_2`
This means that even if you run dshot save, the direction setting is not stored.

See AM32 implementation for reference:
https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/blob/master/Src/dshot.c#L168
Case 7 and 8 are for storing the saveable motor direction and case 20 and 21 are for setting the temporary motor direction.
See this forum post for another data point of evidence to show PX4 is using the wrong commands for setting motor direction:
https://www.rcgroups.com/forums/showpost.php?p=39039010&postcount=2282

This is not an AM32 problem, AM32 appears to be implementing the commands correctly. The `DShot_cmd_spin_direction_reversed` command is used by betaflight for things like turtle mode and motor direction testing. [Here](https://github.com/betaflight/betaflight/issues/10001#issue-653374228) is another point of evidence stating that the '_reversed' command is only for temporary direction change and the other command is for saving the motor direction.

## To Reproduce
On a drone running AM32 ESC firmware:
1. Use commands  `dshot reverse -m 1` and then `dshot save -m 1` as per PX4 motor direction reversal instructions
2. Try testing motor direction - the motor will be reversed properly
3. Reboot the drone (including its ESC)
4. Try testing motor direction again - the motor will have reverted to its original direction rather than being reversed as expected

## Expected behavior
The motor should remain with the set direction (reversed or non reversed), persistent across reboots. 

## Log Files and Screenshots
N/A

## Drone (please complete the following information):
Any drone with AM32 ESC (possibly also blheli but I have not personally tested)

## Additional context
N/A

dagar, 2022-06-02:
@bkueng

bkueng, 2022-06-02:
Good find. Lacking a formal spec I just picked one, and it worked as expected when I tested it.
Can you send a PR to change it?

Note that when using the QGC UI it already uses the other commands: https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/dshot/DShot.cpp#L640=

PR #19758 (2022-06-02): Corrected DShot motor spin direction command - fixes #19756
## Describe problem solved by this pull request
Fixes #19756 - where motor direction command is incorrect (temporary direction change instead of saveable direction change)

## Describe your solution
Replaced temporary direction change DShot commands with saveable DShot commands

## Describe possible alternatives
N/A

## Test data / coverage
Tested on custom flight controller board connected to AM32 ESC

## Additional context
See original issue
closed 2022-06-02 by bkueng