PX4/PX4-Autopilot #19770: Preflight Fail: height estimate not stable

vincentpoont2, 2022-06-07:
## Describe the bug
"Preflight Fail: height estimate not stable" would appear in QGC after bootup. QGC would not display height, it would show --.-- m. 
After removing GPS, the height would show properly. It would go away a couple minute after GPS 3D Fixed.

**Tested using:**
Holybro Pixhawk 4 with PX4 Master (FMUv5)
Also tested other hardware with PX4 Master (FMUv6X, FMUV6C)
Note: FMUv6C May 10th Built did not have this issue.

## To Reproduce
Steps to reproduce the behavior:
1. Power on Flight Controller (with PX4 Master) with GPS 
2. Open QGC, it would show "Preflight Fail: height estimate not stable", Height would display --,-- m
3. Wait a few minute.
4. Error would go away couple minute after GPS 3D fixed.
5. Removed GPS and reboot, this error would not occur. 

## Expected behavior
This error should not show up, height should display correct value. 

## Screenshots
[image]

vincentpoont2, 2022-06-07:
@dagar @bresch, any Idea?

@ryanjAA also have this issue, please comment if you have anything else you like to share.

bresch, 2022-06-07:
FYI @MaEtUgR

MaEtUgR, 2022-06-07:
I had the same on Friday with my bench setup. Pixhawk 4 mini GPS attached but no fix. I noticed that the horizontal velocity value would change ever so slightly when I move the drone by hand the vertical doesn't seem to. Seems like a regression since I haven't changed the setup and it happens immediately. We'll investigate.

MaEtUgR, 2022-06-07:
I tracked this change to cause the problem 🔍 
https://github.com/PX4/PX4-Autopilot/commit/639222dd653c5536e16c02aff6dacf49d2d7159a

MaEtUgR, 2022-06-07:
Here's a lo from boot with this commit: https://logs.px4.io/plot_app?log=006f8348-9c12-4285-a0a3-50c93d04c17b

MaEtUgR, 2022-06-07:
I realized that before this commit my setup switches to Altitude mode before I even connect the ground station. After the commit, I cannot switch to Altitude mode anymore.

MaEtUgR, 2022-06-07:
Here are more logs.
One commit before, drone switches to Altitude mode. Vertical movement reported in QGC:
https://logs.px4.io/plot_app?log=585a7fda-a70c-4391-a8c5-66fdad38681e
With the mentioned commit, drone deosn't switch to Altitude mode. Vertical movement not reported in QGC:
https://logs.px4.io/plot_app?log=ffd970f3-74b0-414c-b751-3aa6753a7dec

MaEtUgR, 2022-06-07:
From discussion, it seems that what's new with this commit is that the innovations of all sensors including the ones that are not currently in use get reported. And the checks then act upon innovations from sensors which are not fused e.g. GPS altitude without a valid fix. As a result the altitude is declared invalid and the preflight checks fail.

closed 2022-06-07 by dagar