PX4/PX4-Autopilot #19783: Java error when 'make px4_sitl jmavsim' ubuntu on macbook pro m1 VM

retoc71586, 2022-06-08:
I am tring to install the PX4 software and just get the simulator to open following the getting started docs. 
At the point of running make px4_sitl jmavsim however i get some java warnings and the build idles forever.

Log:

[0/1] cd /home/parallels/PX4-Autopilot...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/parallels/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /home/parallels/PX4-Autopilot
build_path: /home/parallels/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/parallels/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/parallels/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/parallels/PX4-Autopilot"/test_data
Creating symlink /home/parallels/PX4-Autopilot/build/px4_sitl_default/etc -> /home/parallels/PX4-Autopilot/build/px4_si

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/parallels/PX4-Autopilot/Tools/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 76 source files to /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production/jMAVSim

create_run_jar:
      [jar] Building jar: /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production/jmavsim_run.jar

copy_res:
     [copy] Copying 22 files to /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production

BUILD SUCCESSFUL
Total time: 1 second
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:567)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.UnsatisfiedLinkError: Can't load library: /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production/natives/linux-aarch64//libgluegen_rt.so
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2640)
	at java.base/java.lang.Runtime.load0(Runtime.java:745)
	at java.base/java.lang.System.load(System.java:1873)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibraryInternal(JNILibLoaderBase.java:625)
	at com.jogamp.common.jvm.JNILibLoaderBase.access$000(JNILibLoaderBase.java:64)
	at com.jogamp.common.jvm.JNILibLoaderBase$DefaultAction.loadLibrary(JNILibLoaderBase.java:107)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibrary(JNILibLoaderBase.java:488)
	at com.jogamp.common.os.DynamicLibraryBundle$GlueJNILibLoader.loadLibrary(DynamicLibraryBundle.java:427)
	at com.jogamp.common.os.Platform$1.run(Platform.java:321)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:312)
	at com.jogamp.common.os.Platform.<clinit>(Platform.java:290)
	at com.jogamp.opengl.GLProfile.<clinit>(GLProfile.java:154)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:83)
	at jogamp.opengl.ThreadingImpl$1.run(ThreadingImpl.java:66)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:312)
	at jogamp.opengl.ThreadingImpl.<clinit>(ThreadingImpl.java:66)
	at com.jogamp.opengl.Threading.disableSingleThreading(Threading.java:164)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:129)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:858)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:267)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:194)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
	at me.drton.jmavsim.Simulator.main(Simulator.java:944)
	... 5 more

retoc71586, 2022-06-09:
UPDATE:
make px4_sitl gazebo works for me.
For jmavsim I have tried with all jave versione 8,11,13 but I always get this error ```
Exception in thread "main" java.lang.reflect.InvocationTargetException 
Caused by: java.lang.UnsatisfiedLinkError: Can't load library: /home/parallels/PX4-Autopilot/Tools/jMAVSim/out/production/natives/linux-aarch64//libgluegen_rt.so
```

Bon-1900, 2022-06-28:
same issue on same setup :/

cryptik, 2022-07-04:
Same issues above trying to run jmavsim on M1 native.  I have tried multiple JDKs.  I was able to get the simulator to start using Eclipse Adoptium 17.0.3... but after the sim loads... I get a sequence of these messages

Caught AppContextInfo(Bug 1004) InaccessibleObjectException: Unable to make public static sun.awt.AppContext sun.awt.AppContext.getAppContext() accessible: module java.desktop does not "exports sun.awt" to unnamed module @442675e1 on thread J3D-Renderer-1
    [0]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
    [1]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
    [2]: java.base/java.lang.reflect.Method.checkCanSetAccessible(Method.java:199)
    [3]: java.base/java.lang.reflect.Method.setAccessible(Method.java:193)
    [4]: com.jogamp.nativewindow.awt.AppContextInfo$1$1.run(AppContextInfo.java:40)
    [5]: com.jogamp.common.util.UnsafeUtil.doWithoutIllegalAccessLogger(UnsafeUtil.java:219)
    [6]: com.jogamp.nativewindow.awt.AppContextInfo$1.run(AppContextInfo.java:34)
    [7]: java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
    [8]: com.jogamp.nativewindow.awt.AppContextInfo.<clinit>(AppContextInfo.java:31)
    [9]: com.jogamp.nativewindow.awt.JAWTWindow.<init>(JAWTWindow.java:128)
    [10]: jogamp.nativewindow.jawt.macosx.MacOSXJAWTWindow.<init>(MacOSXJAWTWindow.java:79)
    [11]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
    [12]: java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
    [13]: java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
    [14]: java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
    [15]: java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
    [16]: jogamp.nativewindow.NativeWindowFactoryImpl.getAWTNativeWindow(NativeWindowFactoryImpl.java:105)
    [17]: jogamp.nativewindow.NativeWindowFactoryImpl.getNativeWindowImpl(NativeWindowFactoryImpl.java:66)
    [18]: com.jogamp.nativewindow.NativeWindowFactory.getNativeWindow(NativeWindowFactory.java:654)
    [19]: javax.media.j3d.JoglPipeline$QueryCanvas.addNotify(JoglPipeline.java:8604)
    [20]: java.desktop/java.awt.Container.addNotify(Container.java:2804)
    [21]: java.desktop/java.awt.Window.addNotify(Window.java:791)
    [22]: java.desktop/java.awt.Frame.addNotify(Frame.java:495)
    [23]: java.desktop/java.awt.Window.show(Window.java:1053)
    [24]: java.desktop/java.awt.Component.show(Component.java:1728)
    [25]: java.desktop/java.awt.Component.setVisible(Component.java:1675)
    [26]: java.desktop/java.awt.Window.setVisible(Window.java:1036)
    [27]: javax.media.j3d.JoglPipeline.getBestConfiguration(JoglPipeline.java:8379)
    [28]: javax.media.j3d.Renderer.doWork(Renderer.java:496)
    [29]: javax.media.j3d.J3dThread.run(J3dThread.java:271)

and the simulator won't arm.

julianoes, 2022-07-04:
Have a look if this PR helps: https://github.com/PX4/jMAVSim/pull/132

cryptik, 2022-07-04:
This worked!  Thanks @julianoes!  I am now getting the following exception... 

Caught AppContextInfo(Bug 1004) InaccessibleObjectException: Unable to make public static sun.awt.AppContext sun.awt.AppContext.getAppContext() accessible: module java.desktop does not "exports sun.awt" to unnamed module @10b48321 on thread J3D-Renderer-1
    [0]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
    [1]: java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
    [2]: java.base/java.lang.reflect.Method.checkCanSetAccessible(Method.java:200)
    [3]: java.base/java.lang.reflect.Method.setAccessible(Method.java:194)
    [4]: com.jogamp.nativewindow.awt.AppContextInfo$1$1.run(AppContextInfo.java:40)
    [5]: com.jogamp.common.util.UnsafeUtil.doWithoutIllegalAccessLogger(UnsafeUtil.java:219)
    [6]: com.jogamp.nativewindow.awt.AppContextInfo$1.run(AppContextInfo.java:34)
    [7]: java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
    [8]: com.jogamp.nativewindow.awt.AppContextInfo.<clinit>(AppContextInfo.java:31)
    [9]: com.jogamp.nativewindow.awt.JAWTWindow.<init>(JAWTWindow.java:128)
    [10]: jogamp.nativewindow.jawt.macosx.MacOSXJAWTWindow.<init>(MacOSXJAWTWindow.java:79)
    [11]: java.base/jdk.internal.reflect.DirectConstructorHandleAccessor.newInstance(DirectConstructorHandleAccessor.java:67)
    [12]: java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
    [13]: java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:483)
    [14]: jogamp.nativewindow.NativeWindowFactoryImpl.getAWTNativeWindow(NativeWindowFactoryImpl.java:105)
    [15]: jogamp.nativewindow.NativeWindowFactoryImpl.getNativeWindowImpl(NativeWindowFactoryImpl.java:66)
    [16]: com.jogamp.nativewindow.NativeWindowFactory.getNativeWindow(NativeWindowFactory.java:654)
    [17]: javax.media.j3d.JoglPipeline$QueryCanvas.addNotify(JoglPipeline.java:8604)
    [18]: java.desktop/java.awt.Container.addNotify(Container.java:2804)
    [19]: java.desktop/java.awt.Window.addNotify(Window.java:791)
    [20]: java.desktop/java.awt.Frame.addNotify(Frame.java:495)
    [21]: java.desktop/java.awt.Window.show(Window.java:1053)
    [22]: java.desktop/java.awt.Component.show(Component.java:1729)
    [23]: java.desktop/java.awt.Component.setVisible(Component.java:1676)
    [24]: java.desktop/java.awt.Window.setVisible(Window.java:1036)
    [25]: javax.media.j3d.JoglPipeline.getBestConfiguration(JoglPipeline.java:8379)
    [26]: javax.media.j3d.Renderer.doWork(Renderer.java:496)
    [27]: javax.media.j3d.J3dThread.run(J3dThread.java:271)

and then this one... a few times ....

Caught AppContextInfo(Bug 1004) IllegalAccessException: class com.jogamp.nativewindow.awt.AppContextInfo cannot access class sun.awt.AppContext (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @10b48321 on thread J3D-Renderer-1
    [0]: java.base/jdk.internal.reflect.Reflection.newIllegalAccessException(Reflection.java:394)
    [1]: java.base/java.lang.reflect.AccessibleObject.checkAccess(AccessibleObject.java:674)
    [2]: java.base/java.lang.reflect.Method.invoke(Method.java:568)
    [3]: com.jogamp.nativewindow.awt.AppContextInfo.fetchAppContext(AppContextInfo.java:191)
    [4]: com.jogamp.nativewindow.awt.AppContextInfo.update(AppContextInfo.java:135)
    [5]: com.jogamp.nativewindow.awt.AppContextInfo.<init>(AppContextInfo.java:50)
    [6]: com.jogamp.nativewindow.awt.JAWTWindow.<init>(JAWTWindow.java:128)
    [7]: jogamp.nativewindow.jawt.macosx.MacOSXJAWTWindow.<init>(MacOSXJAWTWindow.java:79)
    [8]: java.base/jdk.internal.reflect.DirectConstructorHandleAccessor.newInstance(DirectConstructorHandleAccessor.java:67)
    [9]: java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
    [10]: java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:483)
    [11]: jogamp.nativewindow.NativeWindowFactoryImpl.getAWTNativeWindow(NativeWindowFactoryImpl.java:105)
    [12]: jogamp.nativewindow.NativeWindowFactoryImpl.getNativeWindowImpl(NativeWindowFactoryImpl.java:66)
    [13]: com.jogamp.nativewindow.NativeWindowFactory.getNativeWindow(NativeWindowFactory.java:654)
    [14]: javax.media.j3d.JoglPipeline$QueryCanvas.addNotify(JoglPipeline.java:8604)
    [15]: java.desktop/java.awt.Container.addNotify(Container.java:2804)
    [16]: java.desktop/java.awt.Window.addNotify(Window.java:791)
    [17]: java.desktop/java.awt.Frame.addNotify(Frame.java:495)
    [18]: java.desktop/java.awt.Window.show(Window.java:1053)
    [19]: java.desktop/java.awt.Component.show(Component.java:1729)
    [20]: java.desktop/java.awt.Component.setVisible(Component.java:1676)
    [21]: java.desktop/java.awt.Window.setVisible(Window.java:1036)
    [22]: javax.media.j3d.JoglPipeline.getBestConfiguration(JoglPipeline.java:8379)
    [23]: javax.media.j3d.Renderer.doWork(Renderer.java:496)
    [24]: javax.media.j3d.J3dThread.run(J3dThread.java:271)

But I am guessing the second one is the reflection error indicated in the troubleshooting guide.  In any case, it runs the simulator and does not appear to cause problems with the operation.  It does have a timeout issue if QGroundControl is not running... but other than that... this PR fixes the running of the simulator on M1.

julianoes, 2022-07-09:
Ah ok, then just ignore that warning.

@ThomasDebrunner should we merge that PR for M1 then?

ThomasDebrunner, 2022-07-10:
Have you tested it? It builds on a RC version of JOGL, I haven't seen any regressions, so I guess we could merge it

julianoes, 2022-07-11:
@cryptik has used it. I don't have an M1 myself.

cryptik, 2022-07-11:
@ThomasDebrunner I was able to get it running on my M1, but I realize now I am using JMavSIM and not Gazebo.  Not sure if that makes a difference?

julianoes, 2022-07-15:
jMAVSim is what we're talking about here. It is based on some Java libraries that were not available for arm64 previously but now are with that PR.

nihavend, 2022-12-18:
What about the future of the PR ?