PX4/PX4-Autopilot #19801: QGC Shows "GPS Disabled" in Sensor Status

vincentpoont2, 2022-06-14:
## Describe the bug
Using Master firmware, flying with GPS in position mode was successful. However it shows "GPS Disabled" on sensor status. Not sure if this is a QGC display issue or a PX4 issue.
[image]

also noted here: https://github.com/PX4/PX4-Autopilot/pull/19774

Flight Log: https://review.px4.io/plot_app?log=5cc286d1-91b7-4662-abcd-f623b50037d8

ryanjAA, 2022-06-15:
I’m seeing the same thing on 4/5x/6c and 6x.

m4v3r, 2022-08-11:
That actually just means that you disabled the GPS checks.
Disabled refers to the checks themselves.
cheers

ryanjAA, 2022-08-11:
That should be notated somewhere or the UI changed since it isn’t clear. 

@hamishwillee thoughts?

TheBinnes, 2022-09-15:
I've not knowingly disabled anything, where do I look in  QGC (or parameters) to re-enable the checks??

amokduke, 2023-02-20:
> I have the same issue. I've not knowingly disabled anything, where do I look in QGC (or parameters) to re-enable the checks??

From https://docs.px4.io/main/en/flying/pre_flight_checks.html
Should be:
COM_ARM_WO_GPS

closed 2025-09-19 by dakejahl