PX4/PX4-Autopilot #19810: Pixhawk 6C - PWM not working

ryanjAA, 2022-06-16:
On master (from today and from a few days back)

When you load up a generic x quadcopter, it arms and there is pwm output when running pwm status on device 0 (main outputs) but nothing spins. It’s not hardware. Plugged in other autopilots and works fine.

Switched over to using the control allocation matrix and still nothing spins when setting outputs to main but if I enable the motors on aux then plug the motors into main or aux it works. 

If wont be sufficient to just test via looking at outputs, physical checks needed and happy to do them. 

Used QGC from today. 

@vincentpoont2 @davids5

vincentpoont2, 2022-07-01:
@dagar any idea?

vincentpoont2, 2022-07-07:
@ryanjAA Can you flash master firmware and try again? Any video or additional detail would be appreciated. We are not seeing this issue in the factory.

ryanjAA, 2022-07-07:
Yes, planning on this tomorrow.

will notate the build currently on it for reference.

ryanjAA, 2022-07-21:
Sorry for not reporting back - tried on the 8th and then with master from today - same thing.

If I plug everything in as expected into the main ports I get no RC output. If I plug into the AUX ports, I get output but when I configure changes (PWM_MAIN3_rev) for instance, that effects the aux 3 port.

We've tried quadcopter and plane.

ryanjAA, 2022-07-22:
The only thing wrong is my ability to read... This works fine. Aux is FMU and IO is main (as usual)... I think the confusion is that on Pixhawk 4 they are on opposite sides (main is left, aux is right) so this works fine and is NOT reversed.