PX4/PX4-Autopilot #19814: Matek H743-Slim V1.5 "System Power unavailable"

PrcsnFlyer, 2022-06-16:
## Describe the bug
The Drone does not pass the Preflight Check. QGC Reports an RC-Receiver Error in the Pre Flight Check Section. 
If you try to arm Via QGC anyways, it gives you an "System Power unavailable Error" 

## To Reproduce
Steps to reproduce the behavior:
1.) Flash FC With a Fresh Bootloader and Firmware Image 
2.) Connect RC Receiver, GPS, ESC etc. 
3.) Power Drone on
4.) Go through Setup(Sensor calibration, RC-Stick calibration etc.) 
5.) Try to Arm via QGC 

## Expected behavior
Motors Should Start spinning 

## Log Files and Screenshots
HW arch: MATEK_H743_SLIM
FW git-hash: 283aad01fdffd39b8b82e58e3433acbdb6ea95c2
FW version: 1.13.0 0 (17629184)
FW git-branch: master
OS: NuttX
OS version: Release 10.2.0 (167903487)
OS git-hash: 91bece51afbe7da9db12e3695cdbb4f4bba4bc83
Build datetime: Jun 11 2022 08:50:40
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
PX4GUID: 00060000000036353839333051170047003b
MCU: STM32H7[4|5]xxx, rev. V
INFO  [param] selected parameter default file /fs/microsd/params
INFO  [param] importing from '/fs/microsd/params'
INFO  [parameters] BSON document size 1561 bytes, decoded 1561 bytes (INT32:29, FLOAT:48)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
Loading airframe: /etc/init.d/airframes/4001_quad_x
INFO  [uavcan] Node ID 1, bitrate 1000000
Board sensors: /etc/init.d/rc.board_sensors
icm42605 #0 on SPI bus 4 rotation 14
mpu6000 #0 on SPI bus 1 rotation 12
dps310 #0 on I2C bus 2 address 0x76
qmc5883l #0 on I2C bus 1 (external) address 0xD
WARN  [SPI_I2C] Already running on bus 1
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
ekf2 [565:237]
Starting Main GPS on /dev/ttyS1
Starting MAVLink on /dev/ttyS5
WARN  [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS5 @ 57600B
INFO  [rc_input] RC scan: SBUS RC input locked
Board extras: /etc/init.d/rc.board_extras
atxxxx #0 on SPI bus 2
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.2.0
nsh> WARN  [commander] Switching to Position is currently not available    
INFO  [commander] Failsafe mode activated    
INFO  [gps] u-blox firmware version: SPG 4.04
INFO  [gps] u-blox protocol version: 32.01
INFO  [gps] u-blox module: NEO-M9N
ERROR [mavlink] FTP: Open failed - out of sessions

INFO  [commander] Failsafe mode deactivated    
ERROR [mavlink] FTP: Open failed - out of sessions

ERROR [uavcan] couldn't start parameter count: -2
ERROR [mavlink] instance 0: RADIO_STATUS timeout
WARN  [mavlink] Session was closed without activity
WARN  [ekf2] primary EKF changed 1 (filter fault) -> 0
INFO  [mavlink] Starting mavlink shell
INFO  [commander] Connection to ground station lost    
INFO  [commander] Data link regained    
WARN  [PreFlightCheck] Preflight Fail: height estimate not stable
ERROR [uavcan] couldn't start parameter count: -2

Add screenshots to help explain your problem.

## Drone:
-Mini TestCopter
-Matek H743-Slim V1 / V1.5 with Matek M9N GPS (Tested both HW-Versions) 

## Additional context
You can also notice, that the FC-PCB Heats up Significantly. Further more the Altitude Estimate Becomes unavailable after some time.

taileron, 2022-06-19:
it seems that the voltage divider is wrong or missing you can try BAT1_V_DIV=11.05 instead of -1

I´m using the mini variant that heats up extremely, cause the small footprint, especially when becs are used for cams or vtx it urgently needs to be cooled in the air

PrcsnFlyer, 2022-06-19:
Thanks for replying! 
The Voltage divider is already set. Voltage and Current Measurement works just fine. It seems like the error is related to `CBRK_SUPPLY_CHK` Circuit breaker. Disabling this one with value 894281 "solved" the issue somehow. Now the drone can be armed.
But now I can notice a huge drift on the yaw axis during movement. I think this is related to the accelerometer and messes up the yaw Fusion with the compass. 
@taileron as you recommended I did all the calibrations on operating temperature. But still there is a lot of noise and drift.

For reference I created a log file on the bench. After reaching the operating temperature I rotated the drone 90° Counter Clockwise - Back - 90° Clockwise - Back, around it's yaw axis. 
You can observe, that the Estimated Yaw angle drifts way further. Also the vibration metrics do not match with the more or less careful movements that i did with the drone. I also disconnected the ESC to reduce overall EMI for the test.

Could this be related to the high clock rate of the CPU? Maybe lowering the clock rate would help with the overall Board Temperature and EMI of the PCB....

Flight log:
https://logs.px4.io/plot_app?log=29970e0b-f221-4e13-99a6-c2ce60ec4759

taileron, 2022-06-20:
You could test with EKF2_MAG_TYPE=none if the drift comes only from the compass (the estimator then starts without physical compass). If you do tests indoors and loudspeakers are in the same room, you need at least 2.5m distance. 
If the system is not powered by lipo but by a power supply, there may be some interference from there as well. At the moment I build the H743 mini in a small 2S Whoop platform, then I will test more precisely which interference factors have the strongest effect or whether a change of the CPU clock multiplier (board.h) would be useful.