PX4/PX4-Autopilot #19821: Desired attitude calculation method

Nyaruko, 2022-06-21:
During the calculation from desired acceleration setpoint to desired body_z direction, it is done by the following code:

```
void PositionControl::_accelerationControl()
{
	// Assume standard acceleration due to gravity in vertical direction for attitude generation
	Vector3f body_z = Vector3f(-_acc_sp(0), -_acc_sp(1), CONSTANTS_ONE_G).normalized();
	ControlMath::limitTilt(body_z, Vector3f(0, 0, 1), _lim_tilt);
	// Scale thrust assuming hover thrust produces standard gravity
	float collective_thrust = _acc_sp(2) * (_hover_thrust / CONSTANTS_ONE_G) - _hover_thrust;
	// Project thrust to planned body attitude
	collective_thrust /= (Vector3f(0, 0, 1).dot(body_z));
	collective_thrust = math::min(collective_thrust, -_lim_thr_min);
	_thr_sp = body_z * collective_thrust;
}
```
It uses a "CONSTANTS_ONE_G" and ignores the _acc_sp(2) when calculating body_z.
May I know what is the advantage of doing this? @MaEtUgR
And is it possible to add a mode that also considers _acc_sp(2) when calculating body_z?

MaEtUgR, 2022-06-22:
@Nyaruko Valid question yes. You're referencing this part: https://github.com/PX4/PX4-Autopilot/blob/479c85047fdee9fcc44d9143dfa2bdaa96040ad0/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L197-L208

I actually explained this in the dev summit 2020. Here's a link to the exact time of the presentation: https://www.youtube.com/watch?v=orvng_11ngQ&t=752s

To summarize, it's to decouple vertical and horizontal position/velocity control because typical multirotor designs where all thrust axes point in body z-axis direction actuator authority allows for much faster dynamics in the body z-axis by quickly increasing and decreasing collective motor thrust compared to the horizontal displacement for which the entire airframe first has to be rotated using differential thrust to align the body z-axis before thrust can be applied in that direction. In simple terms, this strategy causes less very fast unfeasible attitude setpoint changes especially with low thrust e.g. when descending. I know it seems from a purely geometric perspective less correct but this strategy is an easy way to consider the feasibility of setpoint changes with common multirotor dynamics and achieve a more robust tracking of typical trajectories.

> And is it possible to add a mode that also considers _acc_sp(2) when calculating body_z?

We can enable different allocation strategies such that the integrator can choose. This topic actually came up recently in relation to this vehicle for which the mentioned assumptions do not apply: https://github.com/PX4/PX4-Autopilot/pull/19557
Stay tuned for the upcoming dev summit 2022 which starts tomorrow to see more about that 😉

Nyaruko, 2022-06-25:
Thanks for your calrification.

closed 2022-06-25 by Nyaruko