PX4/PX4-Autopilot #19826: Wind Estimator Tuning Equation

Nick-Kakavitsas, 2022-06-22:
I'm trying to figure out where the first half of this equation for predicted acceleration came from.  I understand that the second half is simply the drag force equation simplified to be in terms of the drag acceleration and the ballistic coefficient, but I'm not sure where the (-v * Mcoeff) would come from (where x[0] = Mcoeff = Momentum Drag Coefficient).  I've googled it, read through some seemingly related research papers, but haven't been able to find anything relating a specific Mcoeff to acceleration by multiplying it by the body frame velocity in each direction, as seen here.  Additionally, I've asked an aerospace engineer that I work with and he can't seem to place it either.  I'm not saying that it's wrong, as the tuning seems to work very well in a gazebo simulation, but I just want to know what it's based on for future reference.  

Thanks in advance for any pointing in the right direction.

https://github.com/PX4/PX4-Autopilot/blob/479c85047fdee9fcc44d9143dfa2bdaa96040ad0/src/modules/ekf2/EKF/python/tuning_tools/mc_wind_estimator/mc_wind_estimator_tuning.py#L135-L139

flydevelop, 2022-06-27:
hello.
can u tell me, whats the simplest transfer functions in px4 algorithm?

Nick-Kakavitsas, 2022-06-27:
> hello. can u tell me, whats the simplest transfer functions in px4 algorithm? for example who px4 tune or controle or stabilize a multirotor or plane in presence wind. thank you.

Hi, I'm sorry but I don't think this is the appropriate thread for that question.  My original question was asking about the wind estimation code in the ekf2 implementation, but if you're interested in the control aspects you could open another issue with your question if you want to.  As far as I know, PX4 uses PID control, but I'm not a PX4 developer, I'm just a student interested in wind estimation.  I don't know how they implement their PID, but I'm sure someone could point you in the right direction if you open another issue.  Good luck!

flydevelop, 2022-06-28:
Right. Thank u very much.

good luck.