PX4/PX4-Autopilot #19828: Matek H743-xxx UAVCAN doesn´t start properly and no device detection

taileron, 2022-06-22:
## Describe the bug
the current code doesn´t start uavcan properly

## To Reproduce
Steps to reproduce the behavior:
- connect uavcan device(s) (all pins double checked)
- no device detection
- after uavcan stop and start sometimes it´s detected but no continuous function 

in board.h:
STM32_BOARD_XTAL is only 8MHz on this board maybe this is too less for STM32_FDCANCLK, all other boards use a 24 MHz crystal for it
don´t know how to set CANCLK to 24Mhz  

## Expected behavior
UAVCAN works like expected
(Hereflow is the only small enough device for Whoop sized vehicles)

## Log Files and Screenshots
uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  2 blocks
    Allocated: 2 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      1

CAN1 status:
    HW errors: 31
    IO errors: 31
    RX frames: 0
    TX frames: 1

ESC outputs:
..
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow

Sensor 'rangefinder':
name: uavcan_rangefinder

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 13145 events, 156821us elapsed, 11.93us avg, min 7us max 760us 23.199us rms
uavcan: cycle interval: 13145 events, 3002.73us avg, min 64us max 32936us 659.800us rms

nsh> uavcan stop
nsh> uavcan start
INFO  [uavcan] Node ID 1, bitrate 1000000
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  12 blocks
    Allocated: 12 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   6
    RX transfers:      14
    TX transfers:      4

CAN1 status:
    HW errors: 31
    IO errors: 31
    RX frames: 372
    TX frames: 8

ESC outputs:
..
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow
channel 0: node id 125 --> instance 0

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: node id 125 --> instance 0

Online nodes (Node ID, Health, Mode):
     125 OK         OPERAT    

uavcan: cycle time: 2322 events, 87012us elapsed, 37.47us avg, min 6us max 18543us 615.891us rms
uavcan: cycle interval: 2322 events, 2581.36us avg, min 16us max 18548us 1220.597us rms

## Drone (please complete the following information):
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.

taileron, 2022-06-26:
It has less to do with clock frequencies now.
As soon as UAVCAN_ENABLE=SENSORS+ESC, thereby the standard PWM output is off, then the sensors are recognized and processed continuously. 
Now of course the PWM or dShot output is missing completely. 

-> now it seems that PWM outs and UAVCAN doesn´t work simultaneously

(sensors autom + esc)
uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  37 blocks
    Allocated: 16 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      513
    TX transfers:      5118

CAN1 status:
    HW errors: 352
    IO errors: 360
    RX frames: 1761
    TX frames: 5202

ESC outputs:
..
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow
channel 0: node id 125 --> instance 0

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: node id 125 --> instance 0

Online nodes (Node ID, Health, Mode):
     125 OK         OPERAT    

uavcan: cycle time: 8839 events, 203899us elapsed, 23.07us avg, min 5us max 18561us 297.480us rms
uavcan: cycle interval: 8839 events, 2483.08us avg, min 13us max 32096us 1240.243us rms
nsh>

taileron, 2022-06-28:
found it it´s the io timer no 5 - as soon as its corresponding pwms are disabled uavcan starts to work ...

-> where can I change the timer of uavcan

alinneacsu, 2024-12-17:
> found it it´s the io timer no 5 - as soon as its corresponding pwms are disabled uavcan starts to work ...
> 
> -> where can I change the timer of uavcan

Did you found any solution to this problem ?