PX4/PX4-Autopilot #19831: Mavlink console does not appear to be working

BAI-djohnson, 2022-06-23:
## Describe the bug
I am not able to access the mavlink console in QGroundControl or through the python script in 1.13.0.

## To Reproduce
Steps to reproduce the behavior:
1. Flashed newest firmware (1.13.0)
2. Open QGC
3. Click analyze tools
4. Click console and hit enter
5. No response

## Expected behavior
Expect nsh> to appear.  No response

## Drone (please complete the following information):
- Cube Blue, Standard carrier board

Thank you 
Dane Johnson

Jozmen, 2022-06-23:
I confirm this behavior.

Hardware: Orange Cube (on base board without ADSB, idk whether this has anything to do with this issue)
QGC
PX4: 1.13.0

ecmnet, 2022-06-24:
Might be related to commit https://github.com/PX4/PX4-Autopilot/commit/6d78054f5001548ea2f1d48bb59966e2352bc9fe. I solved this on orangecube  by adding `mavlink start -d /dev/ttyACM0` to `rc.board_mavlink`:

[image]

BAI-djohnson, 2022-06-24:
> Might be related to commit [6d78054](https://github.com/PX4/PX4-Autopilot/commit/6d78054f5001548ea2f1d48bb59966e2352bc9fe). I solved this on orangecube by adding `mavlink start -d /dev/ttyACM0` to `rc.board_mavlink`:

This appears to have fixed my issue.  Thank you.

BAI-djohnson, 2022-06-24:
> Might be related to commit [6d78054](https://github.com/PX4/PX4-Autopilot/commit/6d78054f5001548ea2f1d48bb59966e2352bc9fe). I solved this on orangecube by adding `mavlink start -d /dev/ttyACM0` to `rc.board_mavlink`:

Quick correction.  Once I enabled my app to be compiled with the software the bug persists.

KrzysiekPx4, 2022-07-15:
How can I add mavlink start -d /dev/ttyACM0 to rc.board_mavlink? What is the exact procedure for this operation? I am a beginner user of Orange Cube and I am asking for help.

AlexKlimaj, 2022-07-19:
I have a cube orange and the mavlink shell is not working on 1.13 either.

mrpollo, 2022-07-27:
The mavlink shell isn't working on any platform AFAIK, something is going on with the firmware builds. I know @davids5 was looking into this.

David, are you working on an issue number?

davids5, 2022-07-27:
@mrpollo - No I just verified the alignment bug was not back.

lefamz, 2022-08-24:
@KrzysiekPx4 
Hi, create folder **etc** on sd card from your Cube, create file **extras.txt** inside etc folder (if you don't have it already for other purpose), then just add line **mavlink start -d /dev/ttyACM0** to that file, save and put sd card back in Cube. It should be working now.

david-guardian, 2022-09-28:
This has not resolved the issue for me. QGC version 4.20, and px4 1.13.0. Follow instructions provided above.

david-guardian, 2022-09-28:
This work-around only works on my set up if I boot the cube prior to it's companion computer.

trddo, 2022-11-19:
I have same issue on orange cube v 6.1.7600.16385 using orange cube driver  v2.0.0.10, qgroundcontrol px4  v4.2.4 even after adding mavlink start -d /dev/ttyAMC0 to extras.txt in etc. I am unclear as to how to add to
rc.board_mavlink. 
Would be grateful for further assistance.

sieuwe1, 2023-01-01:
Some issue here with cube orange. I connected my cube orange to 5v with a usb adapter and used a 433mhz telemetry radio. This way the console also worked.

krr27, 2023-05-28:
I have the same issue. It started very recently, perhaps when I switched from v4.2.3 to v4.2.6 a couple of days ago.
I tried the suggestion above (putting the etc/extras.txt file on the SD card).
The problem occurs with two different pix32 V6s, both running FW Release 1.13.3, and OS Release 11.0.0.
The nsh> prompt sometimes appears several minutes after pressing <enter> (but it still doesn't respond to any additional input).
Other QGC functions (e.g. Vehicle Setup) appear to work okay.
Also, I can connect either of the pix32 V6s to a second PC running the same version of QGC, and MAVLink Console appears to work fine.
Both PC's are Windows. The one that has the problem is only about a year old and is running Windows 11. The one that works is a 10-year-old laptop running Windows 10.

krr27, 2023-05-28:
The following additional issues with QGC appeared around the same time, and may be additional clues:
- QGC usually takes a long time to start (e.g., a couple of minutes), whereas it used to start almost instantly
- It frequently crashes, usually when I click the Disconnect button after disconnecting the flight controller (from USB).
- When I attempt to intentionally close the QGC, it often/usually stops responding, and then crashes.
- I also tried running a version that I compiled from source, which I have been using without a problem for a couple of months, and that same problems occur.
- In one case I got the following error message that I haven't seen before:
- Unable to save telemetry log. Error copying to 'C:/Users/krr/Documents/QGroundControl/Telemetry/2023-05-28 13-19-06.6.tlog': 'Failure to write block'.
- I checked that location and there are numerous similar files that were written today.
- I Just now attempted to uninstall QGC (with plans to reinstall it), and the following Windows error:
- An error occurred while trying to uninstall QGroundControl. It may have already been uninstalled.
- I did uninstall QGC v4.2.3 before installing v4.2.6 (a couple of days ago), so perhaps that's when thing went south.

krr27, 2023-05-28:
I just now fixed the issues by doing the following:
- Manually deleted C:\Program Files\QGroundControl, and all other QGC related files that I could find (e.g., C:\Users\krr\AppData\Local\QGroundControl.org, and C:\Users\krr\AppData\Roaming\QGroundControl.org)
- Installed GQC v4.2.3

julianoes, 2023-05-29:
Thanks for the notes. I will close this issue as it should be fixed in `main` and v1.13. If it happens again, please open a new issue, give plenty of information and context and (at) mention me. Thanks.

closed 2023-05-29 by julianoes

jean-dupond, 2023-06-01:
Hello I also have this issue, I want to ask if there is finally a formal solution for this, or if it is about to be fixed in v1.13, thanks.