PX4/PX4-Autopilot #19847: UAVCAN spams MAVLink console output

ecmnet, 2022-06-29:
UAVCAN messages spam mavlink shell as well as ULOG messages in 1.13.0:

[image]

dagar, 2022-06-29:
These are messages coming from the CAN device with node id 125. For the CAN nodes I work with (mainly running PX4) I'm careful to not spam output, but I'm not sure what this is running (hereflow?).

I could add a parameter like `UAVCAN_SUB_LOGMSG` so you could have control over the subscription.

ecmnet, 2022-06-30:
Yes, it’s hereflow.

dagar, 2022-06-30:
https://github.com/PX4/PX4-Autopilot/pull/19856

dagar, 2022-06-30:
Separately maybe we could ask the manufacturer to add a mechanism on the node to silence this output in the future? @proficnc @bugobliterator

bugobliterator, 2022-06-30:
Please follow the steps here https://docs.cubepilot.org/user-guides/here-3/here-3-manual#2.-here-3-here-3+-firmware-upgrade to update the firmware. The update requires either the use of a standalone SLCAN adapter or using CubeOrange+Ardupilot as an SLCAN adapter and connected to Mission Planner.

bugobliterator, 2022-07-01:
@ecmnet @dagar I have updated the firmware for HereFlow to not publish these messages, please update your hardware with the latest firmware that can be installed using following steps: https://docs.cubepilot.org/user-guides/here-3/here-3-manual#2.-here-3-here-3+-firmware-upgrade the software version that should appear after update is <HTML>
</HTML>
</BODY>

<TABLE><TR><TD>1.1.9B687E2</TD></TR></TABLE>
<BODY>

dagar, 2022-07-01:
Thanks @bugobliterator.

On PX4 you can simply place the CAN firmware .bin file on the flight controller SD card and the UAVCAN firmware server will grab it and handle the node upgrade.

ecmnet, 2022-07-03:
@bugobliterator Any direct link to the firmware .bin file?

bugobliterator, 2022-07-03:
> @bugobliterator Any direct link to the firmware .bin file?

https://firmware.cubepilot.org/UAVCAN/

ecmnet, 2022-07-03:
@bugobliterator Thanks a lot for the link. However, either the update did not work (as described above for PX4) or the update worked but did not solve the issue.

bugobliterator, 2022-07-03:
> @bugobliterator Thanks a lot for the link. However, either the update did not work (as described above for PX4) or the update worked but did not solve the issue.

Can you ensure the firmware version is correct 1.1.9B687E2

dagar, 2022-07-03:
Now that I think of I'm not sure if the PX4 UAVCAN firmware server logic that finds the .bin on the SD card works with the ardupilot CAN appdescription.

If I can find some hardware I'll take a look.

ecmnet, 2022-07-03:
As I have no (and cannot install) MissionPlanner, I am limited to the Nutshell console of QGC. `UAVCAN status` gives very little information regarding the FW or the update process currently. I also ran `UAVCAN update` but got no output.

AlexKlimaj, 2022-07-18:
You can use the Dronecan GUI to install the update as well.