PX4/PX4-Autopilot #19852: Mavlink Console Crashes when micrortps_client start (and it doesnt start)

AEGEAN99, 2022-06-30:
## Describe the bug
When I try to connect pixhawk with rtps, I cant start micrortps_client from qground/analize tools nor from the mavlink_shell.py. It crashes and disconnects. After seconds it connects again in qgroundcontrol but micrortps_client still not working.

## To Reproduce
Steps to reproduce the behavior:
1. compiling the firmware with make px4_fmu-v5_rtps upload
2. connected to the qgroundcontrol
3. in qgroundcontrol/analize tools try to start rtps with 'micrortps_client start'
4. terminal stops, when check the connection in qroundcontrol it seems disconnected
5. After seconds qgroundcontrol connects again

AEGEAN99, 2022-07-01:
I solved the question. Microdds caused the problem. 
Solution is to delete the micro_dds folder in the firmware and
Go to PX4-Autopilot/boards/TARGET/rtps.px4board and set CONFIG_MODULES_MICRODDS_CLIENT=n

closed 2022-07-01 by AEGEAN99

frnyb, 2022-09-01:
Has a pull request been opened for this fix?