PX4/PX4-Autopilot #19857: Generic OctoPlane is not supported by Control Allocation Geometry neither by UAVCAN

atikhono1972, 2022-06-30:
## Describe the bug
A clear and concise description of the bug.

1) Control Allocation Geometry in the release 1.13.0 limits the number of motors to 8.
2) Number of ESCs is limited by UAVCAN to 8

Both 1) and 2) dont allow to configure the Standard VTOL in the Octoplane Configuration with UAVCAN ESCs

## To Reproduce
Steps to reproduce the behavior:

- Open GUI of Actuators in the QGroundControl while FMU board is configured with Generic Octoplane VTOL 

## Expected behavior

Number of motors shall be at least 9  in the Control Allocation Geometry
Number of the ESCs shall be at least 9 in the UAVCAN actuators

## Log Files and Screenshots
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- Share the link to the log (Copy and paste the URL of the log)

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.
[image]

phinchey, 2023-04-27:
@bkueng is there any plan to increase the number of actuators in the control allocation to support Octoplanes and others?

bkueng, 2023-04-28:
Yes we have others asking for it as well. The factors that need to be considered are flash size and uorb message size.

phinchey, 2023-05-02:
Ok, I'll be patient, thank you for all your hard work!