PX4/PX4-Autopilot #19862: Position Drifts after 60 seconds when using the Mocap Pose

HaoChen1016, 2022-07-01:
Hi,

I am trying to fuse the Mocap pose into EKF2 instead of VIO pose or GPS data.  The Mocap pose update at 60Hz and it doesn't includes the velocity. For some reason at around 60 seconds+ X and Y begin rapidly diverging and then the EKF stops outputting a horizontal estimate. 

This link includes the details about how I implement it:
https://forum.modalai.com/topic/919/using-motion-capture-systems-optitrack-for-position-estimation/2?_=1656707492656

This link includes the post which has a very similar issue with me:
https://github.com/PX4/PX4-Autopilot/issues/15488

This link includes one of my logged datasets:(I put the quadcopter at the origin but didn't fly it)
https://review.px4.io/plot_app?log=7f27f736-8ea9-4678-80cc-e97801608610

According to the flight review report, there is an error: [ecl/EKF] EV Data drop frame. it seems my external vision data start to drop when EKF2 start to run? Is there anyone know what's the reason for this?
[image]

According to this Doc https://docs.px4.io/master/en/computer_vision/visual_inertial_odometry.html , I set the PX4 parameter MAV_ODOM_LP to 1. Then PX4 will stream back the received external pose as MAVLink ODOMETRY messages. As you can see in the below Figure, the left side is the Odometry which shows quadcopter's pose in NED frame, and the right side is the Mocap pose in the ENU frame, and they match pretty well. Therefore, it seems the EKF2 can receive the Mocap pose.
[image]

Therefore, my guess is that px4 can receive the Mocap pose, but for some reason, the EKF2 cannot integrate it since it will drop?  is there anyone who can give me some suggestion?

Thanks in advance!

gorghino, 2023-04-11:
@HaoChen1016 Did you solve this?

Ali-barari, 2023-07-17:
Hi,
I have a similar issue with my setup. When I switch to Position Hold mode, I get the local estimate error after a few seconds.
I'm sending my LiDAR-SLAM measurements to the Pixhawk 6C (PX4 1.13.3) at 30 Hz. 

1. First of all, The ODOMETRY tab in the QGroundControl Mavlink inspector shows data at 20 Hz and I don't know if that cause an issue or not. What I observed was that time_usec of ODOMETRY were equal to the time_boot_ms of LOCAL_POSITION_NED. But exactly after 55 s, the time_usec of ODOMETRY jumps to a bigger number. I think that's where the problem is coming from.
2. When I start relaying data to Pixhawk, the yaw angle starts at 90 degree (east). How can I fix that?
3. Changes to the PX4_config.yaml file won't be applied when I launch the PX4.launch. I can see the default setting in the parameters. What's the problem?

I followed the instreuctions [here](https://docs.px4.io/main/en/ros/external_position_estimation.html). My companion computer is the Intel NUC which is running a ROS Noetic. Here 's my parameters.
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
[... 245 lines cut ...]
[ INFO] [1689629397.604803388]: VER: 1.1: VID/PID:             3185:0038
[ INFO] [1689629397.604813041]: VER: 1.1: UID:                 3031510a34373938
[ INFO] [1689629397.604840511]: IMU: High resolution IMU detected!
[ INFO] [1689629397.604887517]: IMU: Attitude quaternion IMU detected!
[ WARN] [1689629397.604905506]: CMD: Unexpected command 512, result 0
[ INFO] [1689629397.604926807]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1689629397.605354503]: IMU: High resolution IMU detected!
[ INFO] [1689629397.605384383]: IMU: Attitude quaternion IMU detected!
[ INFO] [1689629397.605442925]: RC_CHANNELS message detected!
[ INFO] [1689629398.614248338]: VER: 1.1: Capabilities         0x000000000000e4ff
[ INFO] [1689629398.614352223]: VER: 1.1: Flight software:     010d03ff (1c8ab2a0d7000000)
[ INFO] [1689629398.614379730]: VER: 1.1: Middleware software: 010d03ff (1c8ab2a0d7000000)
[ INFO] [1689629398.614403637]: VER: 1.1: OS software:         0b0000ff (4a1dd8680cd29f51)
[ INFO] [1689629398.614433779]: VER: 1.1: Board hardware:      00000038
[ INFO] [1689629398.614469072]: VER: 1.1: VID/PID:             3185:0038
[ INFO] [1689629398.614495362]: VER: 1.1: UID:                 3031510a34373938
[ INFO] [1689629407.605248173]: HP: requesting home position
[ INFO] [1689629412.608178624]: GF: mission received
[ INFO] [1689629412.608271749]: RP: mission received
[ INFO] [1689629412.609160944]: WP: mission received
[ INFO] [1689629417.605190412]: HP: requesting home position
[ INFO] [1689629427.605172423]: HP: requesting home position
[ INFO] [1689629437.605209108]: HP: requesting home position
[ INFO] [1689629447.605190030]: HP: requesting home position
[ INFO] [1689629457.605288467]: HP: requesting home position

bresch, 2023-07-18:
>  the yaw angle starts at 90 degree (east). How can I fix that?

What are your PX4 settings? Are you fusing EV yaw as well, or just the position (or velocity)?

Ali-barari, 2023-07-20:
@bresch @mhkabir could you please take a look at the issues I am facing?