PX4/PX4-Autopilot #19863: command denied during calibration: 241

tomzzz989, 2022-07-03:
This problem occurred when I calibrated the magnetic compass. My version is 1.11.0, as shown in the figure
[image]
[image]

dagar, 2022-07-03:
Can you please try v1.13.0 or current PX4 main branch?

tomzzz989, 2022-07-04:
I tried 1.13.0, which will give me another error about SD card. On my other plane, 1.11.0 works normally, so I chose 1.11.0
[image]

dagar, 2022-07-04:
I don't know what the issue is with v1.11.0, but it may have to do with matching QGC version. It definitely wasn't normal to get a "command denied during calibration" error at the time or we would have fixed it. We don't have the resources to maintain old releases, but we can do point releases on v1.13 if there are actual issues.

To silence the "Missing FMU SD card" error you can set `COM_ARM_SDCARD 0` or insert a usable SD card.

tomzzz989, 2022-07-04:
Anyway, thank you for taking the time to think about it。

tomzzz989, 2022-07-05:
I changed the version to 1.10.1 after the problem was solved.

closed 2022-07-05 by tomzzz989