PX4/PX4-Autopilot #19866: Actuator setup for Fixed wing: Change of some default values

jstrebel, 2022-07-04:
The new feature actuator setup is a great help for setting up the mixers. Many thanks to all who have contributed to this. 
Since for FW mostly servos are used that are designed for 50msec. I suggest that the default value 50msec is set.
Reason: Because I left this parameter on default, I burned out a servo.
Furthermore, i suggest, it would be an advantage for FW if the default value in the disarmed state would be set to 1500msec. This would prevent the servo from running to a mechanical endpoint and possibly being overloaded.
Jakob

tstastny, 2022-07-04:
>Furthermore, i suggest, it would be an advantage for FW if the default value in the disarmed state would be set to 1500msec. This would prevent the servo from running to a mechanical endpoint and possibly being overloaded.

@MaEtUgR @dagar wasn't there some discussion of this in the last dev call, is this a current PR in progress?

>Since for FW mostly servos are used that are designed for 50msec. I suggest that the default value 50msec is set.
Reason: Because I left this parameter on default, I burned out a servo.

Sounds fair to me - @sfuhrer @RomanBapst wdyt?

MaEtUgR, 2022-07-04:
> wasn't there some discussion of this in the last dev call, is this a current PR in progress?

A mildly related discussion was about the default failsafe values with the old mixing: https://github.com/PX4/PX4-Autopilot/pull/19836
Otherwise, discussions are about making the new dynamic control allocation default. So it's very useful when we find these corner cases.

> mostly servos are used that are designed for 50msec

I only know servos that run on 50Hz PWM so I'd use that as safe default for any servo.

> Because I left this parameter on default, I burned out a servo

Ouch, I knew this was possible. Usually with widely popular analog servos that do not have error detection on the signal but rely on 50Hz input. We should make sure servos don't accidentally get more than 50Hz PWM if not explicitly configured.

> for FW if the default value in the disarmed state would be set to 1500msec

It has nothing to do with fixed-wings in particular, the ESC on there should not get 1500 PWM by default. It's every servo that should default to 1500 and I thought that's already the case even though I didn't try myself.

FYI @bkueng

jstrebel, 2022-07-05:
@bkueng I understand the priority for the safety considerations. If I look at the attached setup screen for FW then I see 0 Motors. Could this be used for the logic. If 0 motors are configured, then we set the PWM disarmed value to 1500.
BTW: Why is the Message: "One or more actuators still needs to be assigned to an output" ? The attached config works. (The servo for the RC trottle works
[image]
)

sfuhrer, 2022-07-05:
> BTW: Why is the Message: "One or more actuators still needs to be assigned to an output" ? The attached config works. (The servo for the RC trottle works

On the left side you defined 6 servos but then only allocate 4 of them (unless some are on the AUX or UAVCAN).

And  yes I agree with both suggestions here. For adding extra safety we could think about moving the "Propeller removed - enable sliders" to the very top and not allow to make any Setup changes prior it being toggled. As the whole setup really shouldn't be done with props on, not just the actuator testing.

jstrebel, 2022-07-05:
@sfuhrer about the servo allocation. Strange, it works on my FW. I am confused. If I look into QGC Radio Status I see Roll=2,pitch=4, yaw=5, Throttle=1, Flaps=6. Initially I had difficulty to assign throttle and the Flaps to the desired servo. With trying ended up with this. I would be very thankful if you could suggest how the correct setup would look like. Thanks (link with logfile https://review.px4.io/plot_app?log=786c8b38-9a57-45f1-908c-a71cfa247a5e)

MaEtUgR, 2022-07-06:
@jstrebel I'm pretty sure your throttle setup from the screenshot only works for a very limited use case. The autopilot is configured as a glider (0 motors) and the RC throttle stick position is statically forwarded to the MAIN 5 (presumably the ESC). This will run your motor but completely independent of any autopilot control. The motor will always turn according to the throttle stick position independent of the flight mode, any feedback control and I assume even when the autopilot is disarmed.

Suggestion: The idea for a powered plane with a pusher or puller propeller is that you configure it to have one motor and allocate that motor to the pin your ESC is connected to on your setup it seems to be MAIN 5. That way in e.g. Manual and Stabilized mode you can control the throttle manually as soon as the vehicle is armed and in autonomous modes, the throttle is automatically controlled by the autopilot.

MaEtUgR, 2022-07-06:
> is that the risk for that is negligible (as you shouldn't have power connected while configuring anyway)

I'd suggest that the user should get prompted to remove any props before changing this configuration with the explanation that motors can spin up with even just temporary misconfiguration. Power is usually on for me such that I can test the motors.

koebis, 2022-07-07:
@MaEtUgR  Thank you for your help with the actuator setup. I changed the setup as you suggested. screenshot attached. Up to Channel 5 Throttle everthing works as expected. If I test servo 6&7 (Flaps) the servos move. If I try to control the Flaps servos with the Radio, they don't move. Could it be, the choosen profile Bormatec Maja is the culpable?
Please let me know if I should use another place to this issue. I commit that I will contribute this FW setup to Hamish
[image]
[image]

h

koebis, 2022-07-07:
@sfuhrer. Thank you pointing me to this error. I did not realize that Flaps and Elevons are not the same. In the mean time, I did a custom setup for my AAERTF. Everythink seems to work on the bench. Still see a error message. Details in my posting in documentation.

jstrebel, 2022-07-26:
@sfuhrer I like to restart this Request again. The setting in the attached screenshot works. But I still see the error message. If I put the servo number 6 and 7 in the MAIN6 and MAIN7 only one Flap is moving. What I am doing wrong here? Thanks for helping me.
[image]
.

jstrebel, 2022-07-26:
[image]
Works, but error message is still on

PR #19991 (2022-07-31): px4io+pwm_out: set the PWM rate and disarmed value when a channel is first set to a servo
This should simplify the setup a bit.

@jstrebel this adds the logic to set the PWM rate to 50 Hz and disarmed value to 1500 when first setting a servo to one of the PWM outputs.

Fixes #19866
closed 2022-07-31 by dagar