PX4/PX4-Autopilot #19887: when make px4_sitl_default gazebo

747640013, 2022-07-08:
ninja: error: '../../src/lib/events/libevents/events/common.json', needed by 'events/common_with_enums.json', missing and no known rule to make it
Makefile:230: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

747640013, 2022-07-08:
> ninja: error: '../../src/lib/events/libevents/events/common.json', needed by 'events/common_with_enums.json', missing and no known rule to make it Makefile:230: recipe for target 'px4_sitl_default' failed make: *** [px4_sitl_default] Error 1

closed 2022-07-08 by 747640013

liurxing, 2023-05-11:
Hello, how did you solve this problem？

747640013, 2023-05-11:
Mostly due to network issues, the source code was not fully downloaded, or the script did not complete execution entirely.

liurxing, 2023-05-11:
> 

Okay, thanks！

ynyl123, 2025-09-10:
hello,I've downloaded src/lib/events/libevents/events/common.json, but the error still shows that the file is missing, How can I fix this?
ninja: error: 'src/lib/events/libevents/events/common.json', needed by 'events/common_with_enums.json', missing and no known rule to make it
make: *** [Makefile:227：px4_sitl] Error 1