PX4/PX4-Autopilot #19890: Trying out Dynamic control allocatino requires first setting the airframe, then setting SYS_CTRL_ALLOC

junwoo091400, 2022-07-08:
## Describe the bug
After flashing the new firmware, which  makes the control allocation disabled, I must set the *airframe first before enabling control allocation. Otherwise if I first enable control allocation then set the airframe, control allocation gets disabled again.

## To Reproduce
1. Flash new firmware (`make px4_fmu-v5x upload`)
2. Set `SYS_CTRL_ALLOC` to `Enabled` in Parameters Tab (without setting the airframe)
3. Reboot the vehicle (in Parameters tab)
4. Verify that 'Actuators' Tab is present (Control Allocation is enabled)
   [image]
5. Select an airframe and reboot
6. The Actuators tab is gone (and Control allocation is disabled)
   [image]

## Expected behavior
Selecting an airframe shouldn't disable the control allocation

sfuhrer, 2022-07-15:
I don't fully understand why you first set SYS_CTRL_ALLOC and then select the airframe. Note that CA_CTRL_ALLOC is only temporary, and at least for MC we should be able to set it to default very soon.

junwoo091400, 2022-07-21:
Well I would imagine the case being a user wants to test another vehicle with control allocation enabled, so the SYS_CTRL_ALLOC will be Enabled, and the user would switch the airframe.

What confused me was why this parameter would even be touched (e.g. I would expect it to be untouched, like a sensor calibration data for example), is this set to Disabled intentionally in the quadcopter airframe file?

hamishwillee, 2022-07-21:
FWIW I would expect selecting an airframe to reset the vehicle to its defaults. I didn't know sensor data was magically retained.

junwoo091400, 2022-10-21:
Parameter gone, not relevant anymore

closed 2022-10-21 by junwoo091400