PX4/PX4-Autopilot #19901: The Dshot module cannot be started when using dynamic control allocation.

lxcnet4221, 2022-07-12:
## Describe the bug
The 1.13.0 new feature dynamic control allocation works perfectly in PWM output mode, but it seems that Dshot output mode cannot be enabled at the same time as enabling it.

## To Reproduce
1. Set `SYS_CTRL_ALLOC` to 1 and `DSHOT_CONFIG` not to be 0.
[image]
[image]

2. Check status of modules in console.
[image]
It can be seem that the dshot module does not replace the pwm_out module, and the pwm_out module is still running.

## Try to solve
I look into the startup script, and find the param `SYS_CTRL_ALLOC` start up both `pwm_out` and `dshot` modules, which may be conflicting?
https://github.com/PX4/PX4-Autopilot/blob/10a2b4c9f7d34234459b4d3604b99a33491a9d83/ROMFS/px4fmu_common/init.d/rc.interface#L108-L111
I deleted `pwm_out start`, which does disable the `pwm_out` module but the `dshot` module still can't start up.

bkueng, 2022-07-12:
Can you use the [QGC UI](https://docs.px4.io/main/en/config/actuators.html) to configure? In particular, select dshot for the outputs you want: https://docs.px4.io/main/assets/img/qgc_actuators_mc_aux.1a859041.png.
This will then set the right parameters. The `DSHOT_CONFIG` is for the old implementation and will be removed.

lxcnet4221, 2022-07-12:
> Can you use the [QGC UI](https://docs.px4.io/main/en/config/actuators.html) to configure? In particular, select dshot for the outputs you want: [...]

Fixed, and thanks for the quick reply！
Good to know that it is possible to use both PWM and DSHOT protocols at the same time.

closed 2022-07-12 by lxcnet4221