PX4/PX4-Autopilot #19908: Needed NuttX 10.3 for ESP32

andrey-pr, 2022-07-12:
Hello dear friends.
Do you have any plans of integrating NuttX 10.3 to project?
## Describe problem solved by the proposed feature
I plan to port PX4 on ESP32, but I need NuttX 10.3 for it. ESP32 can offer more cheap and small hardware solutions resistant to chip shortage

## Additional context
I found repo with port on github under BSD, but from different author. Now its hard-coded for ESP32. Can I you files from it to contribute this project?

dagar, 2022-07-13:
We're planning to upgrade NuttX in the near future.

andrey-pr, 2022-07-13:
> We're planning to upgrade NuttX in the near future.

Thank you. So, is it allowed to use third-party code licensed under BSD for contributing?

andrey-pr, 2024-10-03:
> @andrey-pr Can I contact you regarding this development. 
> 
> Currently struggling getting to use the spi flash to load the parameters and any help would be appreciated. 

Yes. What messenger do you preffer? Do you have account in Matrix/Element?

andrey-pr, 2024-10-04:
@henrykotze how can I contact you?

henrykotze, 2024-10-06:
@andrey-pr 
Thanks with the follow-up. Somehow I didnt get a email notification.

henrykotze, 2024-10-06:
I dont have account Matrix/Element, but let me know how I can reach you, and I will setup the appropriate account to do so. Otherwise you can reach me on Linkedin and I we can DM there. https://www.linkedin.com/in/henrykotze/