PX4/PX4-Autopilot #19919: Not able to take off from offboard control mode (failsafe mode activated, [PreFlightCheck] mode not suitable for takeoff)

Nandkishor-123, 2022-07-16:
I am trying [this](https://docs.px4.io/main/en/ros/ros2_offboard_control.html) offboard control example. When i run the ros2(foxy) node it says Starting offboard control mode then arm command send and gazebo fails to takeoff. 

[image]

defrag-bambino, 2022-07-19:
In the top terminal run:
`param set COM_RCL_EXCEPT 4
param set NAV_DLL_ACT 0
param set NAV_RCL_ACT 0`

Nandkishor-123, 2022-07-19:
Thank you so much. I ran these commands in the top terminal(where i launched sitl gazebo) and it worked.
Can you explain what was the issue and how you found it? Would be really helpful.

defrag-bambino, 2022-07-20:
> Thank you so much. I ran these commands in the top terminal(where i launched sitl gazebo) and it worked. Can you explain what was the issue and how you found it? Would be really helpful.

You're welcome. I had this issue myself just a few days prior. I learned that these parameters have to be set from someone who has worked with this setup in the past. These essentially define the drone's behavior when there is no RC connected, as is the case in sim. The documentation therefore states somewhere to never use these parameters on a real drone, as it would allow it to fly without a backup RC connected.

Nandkishor-123, 2022-07-20:
Thanks for the information 👍

maxpolzin, 2022-11-01:
Note, `param set NAV_RCL_ACT 0 ` only takes values from 1 to 6 for me with the default being 2.

closed 2025-10-27 by github-actions[bot]