PX4/PX4-Autopilot #19926: Motion Capture Disable Barometer Breaks Estimation

DEADLYBBS, 2022-07-19:
## Describe the bug
When I disable if the system has a barometer, my local_position_NED no longer works. Yaw gets set to 000, and I can no longer view the local position in mavlink inspector.

My Z position drifts and I am under the assumption that it is caused by a noisy barometer reading. X and Y do just fine.

## To Reproduce
Steps to reproduce the behavior:
1. Have motion capture set up as shown in the documentation. I also disable magnetometer sensor fusion as well as the magnetometer and gps in the system parameters.
2. Disable that the system has a barometer. 

## Expected behavior
Disable the onboard sensors (gps, magnetometer, and barometer) to rely on motion capture data. The sensor I disable do not give accurate enough readings indoors.

## Log Files and Screenshots
I can add these and more information as it becomes necessary. Trying to get out quick due to having a meeting.

## Drone (please complete the following information):
- pixracerv1 with esp8266 chip
- optitrack mocap data
- offboard control computer receiving motion capture and passing it to drone via mavros

## Additional context
The drone works perfectly with the barometer, but I have noisy z measurements which I believe causes my drone to not hold a steady z position.

DEADLYBBS, 2022-07-19:
https://discuss.px4.io/t/disable-baro-fusion-in-ekf2/12056

This seems to be a workaround but they have gps data. How could one use the motion capture data instead of the baro data?
Is there a way to adjust the settings in the ekf2 to get further away from trusting the baro measurements, resulting in almost primarily using the mocap data?

dagar, 2022-07-19:
Can you share a log? Can you reproduce this with current stable PX4 v1.13.0 (or the current main branch)?

DEADLYBBS, 2022-07-20:
This is with the current stable PX4 v1.13.0. After some more testing, I think my sensor data may be accurate and the barometer is having little influence over it.
If my data is accurate then the quadcopter will not hold a steady z position. Which may be a tuning issue.
Can the barometer have any affect on the z position when I have the height set to use vision?

DEADLYBBS, 2022-07-28:
https://review.px4.io/plot_app?log=5a0d4421-9e1f-43b5-aee3-bc40eeff4097

Here is a flight log that captures my issue I believe. The issue is not with the EKF2 fusion but with the commanded velocity setpoints. Should I open a new topic and close this issue?

The test is for the drone to fly to -0.5m and hold for 15s. From there it gets a commanded velocity of -0.5m/s. I am also trying to tune the PID for the z velocity controller but the setpoint jumping is a major part of my issue I believe. If I run the python position control example, I do not get the setpoints jumping like this.

DEADLYBBS, 2022-08-01:
The setpoints jumping was a result of bad programming on my part. I accidently made multiple threads that were sending setpoint information even though I was not actively sending setpoint information. My z-axis position data is still very noisy. This is not shown with the Px4 flight review but from when I take my data manually. So there is still an issue with the z-axis position/velocity data.

bresch, 2022-08-02:
When the height source is set to vision, the baro isn't used at all. The barometer is currently a requirement to initialize the filter before it can switch to vision; this is why it won't initialize if the baro is disabled.

bresch, 2022-08-02:
> This is not shown with the Px4 flight review but from when I take my data manually

What do you mean by "take my data manually"? The altitude estimate seems fine