PX4/PX4-Autopilot #19933: gimbal orientation broken in case of pitch == -90 and yaw != 0

svpcom, 2022-07-20:
## Describe the bug
`QGC --[gimbal v1 proto]--> PX4 --[gimbal v1 proto]--> gimbal`
It seems there is a bug with euler angles code which leads to crazy gimbal orientation in case of pitch == -90 and yaw != 0
If I setup these angles  manually (via `gimbal  test yaw -90 pitch -90` ) gimbal orientation is normal.
I've tested that QGC send valid angles (via VEHICLE_CMD_DO_MOUNT_CONTROL).

## To Reproduce
Steps to reproduce the behavior:
1. Connect storm32 gimbal via mavlink to autopilot
1. in QGC set gimbal input as v1 protocol
2. In QGC set gimbal output as v1 protocol
3. In QGC via gimbal control joystick keys set pitch == -90 and yaw == -90 (using QGC via mavlink)
4. gimbal orientation become crazy.

## Expected behavior
Gimbal oriented with pitch -90 and yaw -90

## Drone (please complete the following information):
px4 fmu-v4, release 1.13

svpcom, 2022-07-20:
The following quaternion `0.405580 -0.579228 -0.405580 -0.579228` translates to `0.000000 nan 0.000000` euler angles instead of `-90 0 -110` in `void OutputBase::_calculate_angle_output(const hrt_abstime &t)`

svpcom, 2022-07-21:
It seems that problem is that px4 use PITCH — ROLL — YAW for gimbal instead of ROLL - PITCH - YAW euler rotations. Any ideas how to fix it?

svpcom, 2022-07-21:
[image]
For gimbal unlimited pitch is more practical that unlimited roll because vehicle flight mostly horizontally and roll angle < 90 degrees

julianoes, 2022-07-22:
My snarky comment would be to move to the gimbal v2 protocol which was uses quaternions as much as possible to avoid these problems.

julianoes, 2022-07-22:
What's odd is that yaw does not seem to be taken into account at pitch -90 for gimbal v2.

julianoes, 2022-07-22:
It looks like that's what the mavlink conversion function produces. At pitch -90 deg yaw is left at 0, at pitch -85 deg, everything is correct.

svpcom, 2022-07-22:
But because there are no opensource gimbals, transition to gimbal protocol v2 is not possible. For example STorm32 supports only gimbal v1 protocol.

svpcom, 2022-07-22:
@julianoes Is it possible to change euler rotation sequence in px4 gimbal module?

svpcom, 2022-07-22:
@julianoes 
> It looks like that's what the mavlink conversion function produces. At pitch -90 deg yaw is left at 0, at pitch -85 deg, everything is correct.
with roll-pitch-yaw rotation sequence with pitch == 90 we got aligned roll and yaw axis and gimbal lock (see page 47 in https://www.basecamelectronics.com/files/v3/SimpleBGC_32bit_manual_eng.pdf). So It seems then only way is to change rotation order, but it seems to be hardcoded in `src/lib/matrix/matrix/Euler.hpp`.

svpcom, 2022-07-22:
@julianoes please review pull request

igorsgcampos, 2023-07-19:
I am also experiencing issues when commanding pitch -90, I haven't tested this PR, but intend to to is in near future. When the pitch is -90, the roll actuation seems to be directed to the yaw, please check the gimbal_controls topic of the following topic to see this.
https://review.px4.io/plot_app?log=cc00ed13-c0f3-4c3e-8a51-f430f0451b42