PX4/PX4-Autopilot #19939: Document Control Allocation concept for Developers

junwoo091400, 2022-07-21:
## Background
There were [some interest](https://discuss.px4.io/t/px4-dev-call-july-20-2022/28275#q2-custom-multicopter-control-using-dynamic-control-allocation-salvatore-9) in a documentation of how to write your custom Control Allocation Actuator effectiveness matrix & create a custom controller out of it, which I think is quite important detail that users / researchers should have access to.

Also, the Control Allocation pipeline isn't so straightforward for a regular developer / user, as it requires some understanding in math and control theory (matrix part). So it would be nice to document exactly how the `vehicle_torque_setpoint`, `actuator_outputs`, `actuator_controls`, etc are related to each other.

## Describe your preferred solution
1. Document the whole control allocation concept in theoretical level (Actuator Effectiveness matrix, Matrix inverse)
2. Document how control allocation pipeline is implemented in PX4 (All the meanings of the related uORB topics, which topics gets published where and gets processed where, and where the matrix inversion occurs, etc)
3. Document how to create your own custom Control Allocation model (Using a simple example, like any vehicle that isn't supported yet - @sfuhrer any ideas?)

## Additional context
There's already quite some [nice improvements](https://github.com/PX4/PX4-user_guide/pull/1962) coming in documenting how to setup / use control allocation, we just need a more in-depth documentation on how it works so that we can make developers and researchers' life easier!

junwoo091400, 2022-10-21:
Basic support was added, implementing Control Allocation model should be done by referring to the latest code changes, so that part is unnecessary.

https://docs.px4.io/main/en/concept/control_allocation.html
https://github.com/PX4/PX4-user_guide/pull/1923

closed 2022-10-21 by junwoo091400

sfuhrer, 2025-10-20:
Does https://www.youtube.com/watch?v=NQwepGLMzZ4 help you?

sfuhrer, 2025-10-20:
Would be great indeed, but doesn't exist afaik. If you could compose such a document, me and all our future researchers would be extremely appreciative!