PX4/PX4-Autopilot #19945: Insufficient Space for FLASH on Pixhawk 6x build

huiyulhy, 2022-07-21:
## Describe the bug
Tried to compile code in the latest version of main for a Pixhawk 6x, but ran into an error during build that there was insufficient space in the FLASH region

## To Reproduce
Steps to reproduce the behavior:
1. git clone latest main branch (does not need to include submodules)
2. run build command: `make px4_fmu-v6x_default`
3. See error on build command

## Expected behavior
Should run into error compiling that FLASH region overflowed. 
[image]

Will0fD, 2022-07-21:
Same here. 
[image]
`git log`
[image]

muramura, 2022-07-21:
Commit hash (main): 32c6ec06
This commit was successfully built.

[image]

Will0fD, 2022-07-21:
> Commit hash (main): [32c6ec0](https://github.com/PX4/PX4-Autopilot/commit/32c6ec061e46785b9913ae22409fd41f537987da) This commit was successfully built.
> 
> [image]

Your FLASH `Used Size` is about 20KB less than ours. 🤔 Any idea why? We're on the same commit. Maybe it's the compiler?
[image]

huiyulhy, 2022-07-21:
> Commit hash (main): [32c6ec0](https://github.com/PX4/PX4-Autopilot/commit/32c6ec061e46785b9913ae22409fd41f537987da) This commit was successfully built.
> 
> [image]

Just curious, what version of Linux are you running on for your development environment?

muramura, 2022-07-22:
master/main (6a0f394d)
[image]

Will0fD, 2022-07-22:
main (6a0f394) :(
[image]

muramura, 2022-07-22:
I use the following version of my GCC.
ArduPilot uses this version.
I also develop ArduPilot.

gcc-arm-none-eabi-10-2020-q4-major

Aeolus-EricKhoo, 2022-07-27:
I met the same issue as well, what I did is using the use the command make `px4_fmu-v6x boardguiconfig` and remove some of the unused modules or drivers (for example uavcan, camera_capture, and camera_trigger), then it solved the problem.

junwoo091400, 2022-12-02:
It would be interesting to compare the compiler differences (20KB result is quite big!), but this itself is a re occurring problem in PX4 👻