PX4/PX4-Autopilot #19946: RCIN on PIXHAWK 6X does not receive channel data from receiver

muramura, 2022-07-21:
## Describe the bug

V1.13.0 release version was written to Holybro Pixhawk 6X.
This version does not receive data from FUTABA R3008SB connected to RCIN.

## To Reproduce

1. write V1.13.0 release version to Holybro Pixhawk 6X
2. look at the QGC radio calibration screen.

## Expected behavior

What QGC can do for radio calibration

## Log Files and Screenshots

None

## Drone (please complete the following information):

Frame is a Holybro X500 V2
FC is a PIXHAWK 6X

## Additional context

 In the same environment, the latest release of ArduPilot was written by QGC.
The status of each channel was displayed on the radio calibration screen.
Operated the radio sticks and switches and changed normally.

muramura, 2022-07-22:
I contacted Holybro.
I received a response from Holybro.
"Please build master/main and update the firmware."
I did this.
I was able to calibrate the radio.

[image]

1042829677, 2022-07-27:
hello, I have the same problem。
"Please build master/main and update the firmware."
What does “the firmware” mean？