PX4/PX4-Autopilot #19956: [Project tracker] Rover control improvements

junwoo091400, 2022-07-25:
## About
Current Rover Position Control's logic has a number of weird logic and is generally messy. This is a project tracker to cleanup & improve rover control.

### Related branch / PRs
While testing integration of Racing Boat with PX4, I have been refactoring the Rover code quite a bit in 
https://github.com/junwoo091400/PX4-Autopilot/tree/boat_bigstorm. And my plan is to squash these commits and do a refactor PR with improvements applied from this Issue tracker.

### TODOs
- [ ] Make Rover position / velocity / attitude / rate controller cascade each other
- [ ] Remove legacy fixed-wing related code in rover controller
- [ ] 

## Problems
### Non-cascaded controller
Currently in Rover Position Control module, each controller (position / velocity / attitude / rate) runs on it's own setpoint -> actuator output pipeline. This in my opinion is bad since we are disregarding all the PID values tuned for the rate controller and we are directly setting the throttle & yaw control in high-level (position control for example). Here are some examples:

Position control:
https://github.com/PX4/PX4-Autopilot/blob/8ca28f3796cdf4c903c14f180045ba1b85760ab2/src/modules/rover_pos_control/RoverPositionControl.cpp#L297-L304

Velocity control:
https://github.com/PX4/PX4-Autopilot/blob/8ca28f3796cdf4c903c14f180045ba1b85760ab2/src/modules/rover_pos_control/RoverPositionControl.cpp#L364-L368

[image]

What I think should be the case is a cascaded controller where each level of controller gives setpoint to the lower level controller, and in the end the rate controller should be the sole part of the code that sets the actuator output.

https://docs.px4.io/main/en/flight_stack/controller_diagrams.html#multicopter-control-architecture

@Jaeyoung-Lim what do you think?

### Lots of if / else conditions for checking which 'mode' we are in
There seems to be a lot of if / else statements to manually distinguish and guide code paths so that we can control position / velocity / attitude etc in a correct fashion, like this:

https://github.com/PX4/PX4-Autopilot/blob/8ca28f3796cdf4c903c14f180045ba1b85760ab2/src/modules/rover_pos_control/RoverPositionControl.cpp#L487-L499

But I think this is quite hard to read. I haven't checked how the multicopter deals with this, but could there be a better way to do this?

Jaeyoung-Lim, 2022-07-25:
> Currently in Rover Position Control module, each controller (position / velocity / attitude / rate) runs on it's own setpoint -> actuator output pipeline.

@junwoo091400 The L1 control logic is based on acceleration control and therefore a cascaded control structure (where position is controlled by velocity setpoints) are not relevant for using an L1 guidance logic.

The velocity controller was added in https://github.com/PX4/PX4-Autopilot/pull/13225 to support offboard velocity setpoints, and was not intended to be used for position control. 

> This in my opinion is bad since we are disregarding all the PID values tuned for the rate controller and we are directly setting the throttle & yaw control in high-level (position control for example).

This I agree, and is simply because we are missing a lower level rate controller. This was why I tried to add https://github.com/PX4/PX4-Autopilot/pull/18317 .

I would claim that the biggest missing piece for boats are the low level rate controller. This is because unlike rovers boats have a low damping on yaw and therefore actuator inputs do not map directly to yaw rates. For normal rovers this assumption is not too bad and that is why it is was working okay with the current formulation on rovers. IMHO, we should prioritize on getting the rate controller for rovers (boats) reliable before we add any improvements for the position controller.

> There seems to be a lot of if / else statements to manually distinguish and guide code paths so that we can control position / velocity / attitude etc in a correct fashion, like this

This is how it is done for multicopters, fixed wings also. Feel free to improve this, but I would not relate this to improving rover. This is related to how commander communicates control modes to the controllers.

junwoo091400, 2022-07-25:
> This is because unlike rovers boats have a low damping on yaw and therefore actuator inputs do not map directly to yaw rates

What does this mean? I am lacking some understanding here I think :thinking: 

Also, regarding improving the rate control, how would taking into account the velocity come into play? I am thinking of a case for example when a user tries to do yaw rate control while stationary, and it would saturate the I component, and then would apply this constant when the user then throttles up and starts moving the boat.

Jaeyoung-Lim, 2022-07-25:
> This is because unlike rovers boats have a low damping on yaw and therefore actuator inputs do not map directly to yaw rates
>  What does this mean? I am lacking some understanding here I think 

If you apply some yaw torque to the vehicle, rovers normally converge to a specific yaw rate very quickly since the ground damping forces are quire significant. For a boat, since water provides much less significant damping effect on the vehicle the vehicle might have more complicated dynamics on yaw

> Also, regarding improving the rate control, how would taking into account the velocity come into play? I am thinking of a case for example when a user tries to do yaw rate control while stationary, [...]

This needs scaling the actuator effectiveness depending on speed. This may be different whether it is a differential drive vehicle or a ackerman steering vehicle. You can probably use insights from fixed wings relative to airspeed.

junwoo091400, 2022-11-11:
Don't think we really need this Issue :+1:, we can discuss everything in separate baby step PRs

closed 2022-11-11 by junwoo091400