PX4/PX4-Autopilot #19960: libprotobuf ERROR  file already exists in database: Imu.proto

Tina-Lau, 2022-07-26:
when I execute _make px4_sitl gazebo_, something went wrong. At first I didn't see any words in red that told what's wrong, until I rolled back and discovered something concerned _libprotobuf_ in white but seems fatal. I attached the log in terminal and highlighted the messages in bold(right above PX4 logo). I couldn't find anything similar on the Internet, can someone help?

-- PX4 version: v1.13.0-alpha1
-- PX4 config file: /home/lam/px4/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-569-g71fc1b8
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lam/px4/PX4-Autopilot/build/px4_sitl_default
[0/817] git submodule src/drivers/gps/devices
[1/817] git submodule src/lib/ecl
[2/817] git submodule mavlink/include/mavlink/v2.0
[7/817] git submodule Tools/sitl_gazebo
[12/817] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
[... 200 lines cut ...]
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
Gazebo multi-robot simulator, version 9.19.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:135] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo
^CERROR [px4] Startup script returned with return value: 2
/home/lam/px4/PX4-Autopilot/Tools/sitl_run.sh: line 235: 12642 Aborted                 (core dumped) gzserver $verbose $world_path  (wd: ~/px4/PX4-Autopilot/build/px4_sitl_default/tmp)
/home/lam/px4/PX4-Autopilot/Tools/sitl_run.sh: line 244: kill: (12642) - No such process
/home/lam/px4/PX4-Autopilot/Tools/sitl_run.sh: line 246: kill: (12705) - No such process
ninja: build stopped: interrupted by user.
Makefile:231: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Interrupt

AlexWUrobot, 2023-07-12:
I have a similar issue, when running 
`roslaunch gazebo_ros empty_world.launch world_name:=$(find vrx_gazebo)/worlds/example_course.world`
and then import
`rosrun gazebo_ros spawn_model -sdf -file $(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/iris/iris.sdf -model iris -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0`
[image]

fgsdfgsda, 2023-09-27:
I have a similar issue, did you  solve it?
[image]