PX4/PX4-Autopilot #19961: example folder in build folder only contains fake gps

amoogeona11, 2022-07-26:
## Environments
Ubuntu 20.04
ROS Noetic
python3.8.10
px4-autopilot v1.13.0
Pixhawk 4

## Describe the bug
I'm trying to read imu data from Pixhawk4 using ROS and I'm stuck in this problem.

According to the tutorials in docs.px4.io, I used the command `make px4_fmu-v5_default upload` to compile and upload but when I entered nsh and typed help to see what's inside, I couldn't find any files in examples folder except fake_gps. I tried changing Kconfig and CMakelists in other examples just like the ones in fake_gps but it didn't work.

I also tried putting px4_simple_app in modules folder and fixed Kconfig and CMakelists files but it didn't work either. After compiling, build/px4_fmu-v5_default/src/example folder only contains fake_gps.

I tried older versions such as v1.10.0 in ubuntu 18.04 which had no problem. 
It seems like a bug...

[image]

Jaeyoung-Lim, 2022-07-26:
@amoogeona11 Could you share the whole build log? Which version of PX4 were you using?

amoogeona11, 2022-07-26:
Thanks for the comment!

seok@seok:~/PX4-Autopilot$ make px4_fmu-v5_default
-- PX4 version: v1.13.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/seok/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- drivers/px4io: ROMFS including px4_io-v2_default
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/seok/PX4-Autopilot/build/px4_fmu-v5_default
[0/1156] git submodule src/drivers/gps/devices
[7/1156] git submodule platforms/nuttx/NuttX/nuttx
[12/1156] git submodule src/drivers/uavcan/libuavcan
[13/1156] git submodule src/modules/mavlink/mavlink
[14/1156] Generating Mavlink development: src/modul...nk/mavlink/message_definitions/v1.0/development.xml
Validating /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /home/seok/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 229 MAVLink message types in 4 XML files
Generating C implementation in directory /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/development
Generating C implementation in directory /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/standard
Generating C implementation in directory /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/common
Generating C implementation in directory /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/mavlink
[1154/1156] Linking CXX executable px4_fmu-v5_default.elf
Memory region         Used Size  Region Size  %age Used
      FLASH_ITCM:          0 GB      2016 KB      0.00%
      FLASH_AXIM:     1967049 B      2016 KB     95.29%
        ITCM_RAM:          0 GB        16 KB      0.00%
        DTCM_RAM:          0 GB       128 KB      0.00%
           SRAM1:       45480 B       368 KB     12.07%
           SRAM2:          0 GB        16 KB      0.00%
[1156/1156] Creating /home/seok/PX4-Autopilot/build/px4_fmu-v5_default/px4_fmu-v5_default.px4

current branch is v1.13.0
main branch brings the same result

Jaeyoung-Lim, 2022-07-26:
@amoogeona11 Did you include `px4_simple_app` into the `px4-fmu_v5_default build target? 

I don't think the `px4_simple_app` is included by default and it is expected that you don't see it on the default build target

amoogeona11, 2022-07-26:
Where can I check the build target?
I cloned PX4-Autopilot directly through terminal and I didn't fix anything. (I tried fixing Cmakelists and Kconfig tho)
I see the examples folder is not included from the build log but it's weird that only the fake_gps in examples folder is built.

Jaeyoung-Lim, 2022-07-26:
@amoogeona11 https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v5/default.px4board

amoogeona11, 2022-07-26:
Now it works!!
Thanks for your comments. I really appreciate it.

closed 2022-07-26 by amoogeona11