PX4/PX4-Autopilot #19970: [Project Tracker] Sensor configuration display UI

junwoo091400, 2022-07-27:
## Describe problem solved by the proposed feature
Currently it is hard to figure out which sensors are connected where, with which configuration and which priority (e.g. `CAL_MAG0_PRIO`) in a single view. For example during my [boat project](https://discuss.px4.io/t/rc-speed-boat-with-px4-episode-1-using-px4-to-control-the-boat/28429#things-that-are-still-ambiguous-to-me-22) build, I wasn't sure if I needed to set the priority of magnetometer manually or not.

This is also quite related to the calibration pain addressed in the issue #19459. What can we do to improve user experience for setting up the sensors? If so, which part should be addressed and how?

What could be great (in my opinion) could be something like this in QGC:

<mag-type> | <mag-orientation> | <mag-priority> | <mag-protocol?> # First mag sensor
<mag-type> | <mag-orientation> | <mag-priority> | <mag-protocol?> # One for the second mag
<baro-type> |                               | <baro-priority> | <baro-protocol>
<gyro-type> | <gyro-orientation> | <gyro-priority> | <gyro-protocol>

But I am also not sure what others expectation / needs are, so any feedback would be appreciated!

## Describe your preferred solution
* @davids5 pointed out that transferring device tree information over to QGC could be a solution. Could you elaborate on that?
* @dagar pointed out that having a unified view in general would be good. Could you let me know if a view I suggested above is similar to what you have imagined?

junwoo091400, 2022-10-21:
@Igor-Misic I believe you were working with Device Trees, do they exist in NuttX yet?

Igor-Misic, 2022-10-21:
> @Igor-Misic I believe you were working with Device Trees, do they exist in NuttX yet?

As far as I know, there were just some ideas and no one worked at device tree yet.

JugalIwPtl, 2024-06-22:
/attempt #19970 On it !!

RS-labhub, 2024-11-10:
Why don't we make it simply like the below??

```
| **Sensor Type** | **Orientation**      | **Priority**   | **Protocol**       | **Status**                 |
|-----------------|----------------------|----------------|--------------------|----------------------------|
| Magnetometer #1 | XYZ (default)        | Auto (Highest) | I2C                | ✅ Connected               |
| Magnetometer #2 | YZX (custom)         | 2nd Priority   | SPI                | ⚠️ Adjust Priority         |
| Barometer       | N/A                  | Auto           | I2C                | ✅ Calibrated              |
| Gyroscope #1    | XZY (custom)         | Primary        | CAN                | ✅ Connected               |
| Gyroscope #2    | ZYX (inverted)       | Secondary      | CAN                | ⚠️ Needs Calibration       |
```

(I built this table using chatGPT by proposing my solution)

junwoo091400, 2024-12-11:
> @junwoo091400 is this still valid for a bounty? cus it shows so on the algora portal. if yes, then i would be interested in taking this up .

I was trying to test the system so it wasnt a real bounty, but hey if you actually build one, I wouldn't mind giving out $10!

IIITManjeet, 2024-12-30:
sure

bitdeckai, 2026-02-14:
I am also picking this low-bounty issue in parallel. I will propose an implementation plan and PR link after local validation.