PX4/PX4-Autopilot #19974: SITL Fixedwing mission can't land without range sensor

julianoes, 2022-07-27:
## Describe the bug

I can no longer land at the end of a mission in SITL using the `plane` model.

The warning is:
> WARN  [fw_pos_control_l1] Landing aborted: terrain estimate not found	

Presumably this is because there is no distance sensor configured for the plane which is now required.

It would be good to flag such a mission as invalid if there is no range sensor in the preflight check.
Also, the plane SITL model should have a range sensor.

## To Reproduce
Steps to reproduce the behavior:
1. `make px4_sitl gazebo_plane`
2. Plan fixedwing mission with takeoff and 
3. Fly! And keep flying...

tstastny, 2022-07-29:
>It would be good to flag such a mission as invalid if there is no range sensor in the preflight check.

I can take a look at this.

>Also, the plane SITL model should have a range sensor.

@Jaeyoung-Lim is this easy enough to implement?

Jaeyoung-Lim, 2022-07-29:
> Also, the plane SITL model should have a range sensor.

@tstastny Yes, actually promised @ryanjAA to do this few weeks ago :smile: I will do it within this week

@julianoes @tstastny Does this mean we need a distance sensor in the default fixed wing model? I was planning to add a model called `plane_lidar`, but if we want to add it to the default model this is fine on myside

alelse, 2023-04-25:
Is there a fix for this yet or a work around?