PX4/PX4-Autopilot #19986: If kakute H7 can use EKF2?

ckl677, 2022-07-29:
I want to know if kakute H7 can use EKF2

bresch, 2022-08-02:
Hi, yes, but it's not enabled by default on this board since it doesn't have a mag (which isn't an issue if you don't use GPS). You can enable it by setting [SYS_MC_EST_GROUP](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SYS_MC_EST_GROUP) to 2 (EKF2)

ckl677, 2022-08-04:
> Hi, yes, but it's not enabled by default on this board since it doesn't have a mag (which isn't an issue if you don't use GPS). You can enable it by setting [SYS_MC_EST_GROUP](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SYS_MC_EST_GROUP) to 2 (EKF2)
Thanks,I set SYS_MC_EST_GROUP to 2,but I can not arm, if I use Q estimator, I can arm,what is the problem?

ckl677, 2022-08-04:
> > Hi, yes, but it's not enabled by default on this board since it doesn't have a mag (which isn't an issue if you don't use GPS). You can enable it by setting [SYS_MC_EST_GROUP](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SYS_MC_EST_GROUP) to 2 (EKF2)
> 
> Thanks,I set SYS_MC_EST_GROUP to 2,because I want to use vision,so I set EKF2_AID_MASK=24,EKF2_HGT_MODE=Vision,but I can not arm in stabilized mode, if I use Q estimator, I can arm in stabilized mode,what is the problem?Could I use vision in kakute H7?

Use EKF2, I can arm in stabilized mode now,I set EKF2_MAG_TYPE=None.Now I want to know if I can use vision in kakute H7.

bresch, 2022-08-08:
> Now I want to know if I can use vision in kakute H7

Yes, if you have a valid vision data source, EKF2 should initialize its heading with the EV yaw fusion and then start EV pos fusion.

w407022008, 2022-08-08:
Hello @ckl677 , I am very happy to hear that you are using the kakute h7 flight control, I am planning to buy it to replace my pixhawk4. may I ask if you are using the h7 or the h7 mini and how do you feel with it? Can we easily compile and upload px4 programs like we do with the pixhawk series? As far as I know it uses the betaflight firmware by default.

DEADLYBBS, 2022-11-15:
I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. 
The kakute h7 will not arm, more specifically, it states that AHRS is disables and that there is a Pre-Arm Check error. How can I view what is causing this error? I have tried accessing the flight logs but that does not seem to provide any information. 
https://logs.px4.io/plot_app?log=02ade54d-d83b-41c1-b304-6ce31e9ff680

Any help would be much appreciated!

ckl677, 2022-11-15:
> Hello @ckl677 , I am very happy to hear that you are using the kakute h7 flight control, I am planning to buy it to replace my pixhawk4. may I ask if you are using the h7 or the h7 mini and how do [...]

Yes,it is easy to use,you can upload programs in ubuntu.But it is not very well compared with pixhawk4.It performs worse than pixhawk4 in the room using vision.

ckl677, 2022-11-15:
> I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. The kakute h7 will not arm, more specifically, it states that [...]

Because kakuteh7 does not have a compass, it is necessary to set ekf to disable the compass and use the external heading angle for positioning.

DEADLYBBS, 2022-11-15:
> > I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. The kakute h7 will not arm, more specifically, it states that AHRS is disables and that there is a Pre-Arm Check error. How can I view what is causing this error? I have tried accessing the flight logs but that does not seem to provide any information. https://logs.px4.io/plot_app?log=02ade54d-d83b-41c1-b304-6ce31e9ff680
> > Any help would be much appreciated!
> 
> Because kakuteh7 does not have a mag, it is necessary to set ekf to disable the mag and use the external heading angle for positioning.

I have disabled that. With it disabled, will it never be able to arm unless I get an external heading angle?
I am planning on using optitrack motion capture software and that is what I used with the PixRacer instead of GPS.
Is there any information that would show up in the pre-flight checks showing the lack of any sensors?

ckl677, 2022-11-15:
> > > I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. The kakute h7 will not arm, more specifically, it states that AHRS is disables and that there is a Pre-Arm Check error. How can I view what is causing this error? I have tried accessing the flight logs but that does not seem to provide any information. https://logs.px4.io/plot_app?log=02ade54d-d83b-41c1-b304-6ce31e9ff680
> > > Any help would be much appreciated!
> > 
> > 
> > Because kakuteh7 does not have a mag, it is necessary to set ekf to disable the mag and use the external heading angle for positioning.
> 
> I have disabled that. With it disabled, will it never be able to arm unless I get an external heading angle? I am planning on using optitrack motion capture software and that is what I used with the PixRacer instead of GPS. Is there any information that would show up in the pre-flight checks showing the lack of any sensors?
> 
> Edit: Further reading shows that a compass is necessary. How difficult would it be to add a magnetometer? I have some generic ones lying around.

Can you arm in stabilize mode using ekf2?I use it with motion capture too.You can use yaw information from motion capture instead of mag.

DEADLYBBS, 2022-11-15:
> Can you arm in stabilize mode using ekf2?I use it with motion capture too. You can use yaw information from motion capture instead of mag.

I cannot currently, probably due to lack of magnetometer.  I need to connect an esp8266-01 wifi module for the motion capture information. Unless there is an easier way to get mocap to the board. 
I know how to get the yaw information and set up everything for motion capture and autonomous flights, I hope. I did this before with a PixRacer and had tremendous success, just the drone was rather heavy and not able to hold the Z position. I am glad that someone has gotten this working! Hopefully I can get everything working for a demonstration I have to give Thursday!

bresch, 2022-11-15:
@DEADLYBBS Did you set:
- [SYS_HAS_MAG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SYS_HAS_MAG) to 0 and
- [EKF2_MAG_TYPE](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_MAG_TYPE) to 5 (none)

ckl677, 2022-11-15:
> I cannot currently, probably due to lack of magnetometer. I need to connect an esp8266-01 wifi module for the motion capture information. Unless there is an easier way to get mocap to the board. I [...]

To get the information of motion capture,I use a offboard computer to receive the information and use mavros via telem2.To use telem2,you should change the compile file.

ckl677, 2022-11-15:
> To get the information of motion capture,I use a offboard computer to receive the information and use mavros via telem2.To use telem2,you should change the compile [...]

> > > > > > I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. The kakute h7 will not arm, more specifically, it states that AHRS is disables and that there is a Pre-Arm Check error. How can I view what is causing this error? I have tried accessing the flight logs but that does not seem to provide any information. https://logs.px4.io/plot_app?log=02ade54d-d83b-41c1-b304-6ce31e9ff680
> > > > > > Any help would be much appreciated!
> > > > > 
> > > > > 
> > > > > Because kakuteh7 does not have a mag, it is necessary to set ekf to disable the mag and use the external heading angle for positioning.
> > > > 
> > > > 
> > > > I have disabled that. With it disabled, will it never be able to arm unless I get an external heading angle? I am planning on using optitrack motion capture software and that is what I used with the PixRacer instead of GPS. Is there any information that would show up in the pre-flight checks showing the lack of any sensors?
> > > > Edit: Further reading shows that a compass is necessary. How difficult would it be to add a magnetometer? I have some generic ones lying around.
> > > 
> > > 
> > > Can you arm in stabilize mode using ekf2?I use it with motion capture too. You can use yaw information from motion capture instead of mag.
> > 
> > 
> > I cannot currently, probably due to lack of magnetometer. I need to connect an esp8266-01 wifi module for the motion capture information. Unless there is an easier way to get mocap to the board. I know how to get the yaw information and set up everything for motion capture and autonomous flights, I hope. I did this before with a PixRacer and had tremendous success, just the drone was rather heavy and not able to hold the Z position. I am glad that someone has gotten this working! Hopefully I can get everything working for a demonstration I have to give Thursday!
> 
> To get the information of motion capture,I use a offboard computer to receive the information and use mavros via telem2.To use telem2,you should change the compile file,because it doesn't have /dev/ttyS1.
The file is 

> > > > > > I cannot get the ekf2 to work. I have experience with the original PixRacer board and have gotten it to fly with the ekf2 without gps. The kakute h7 will not arm, more specifically, it states that AHRS is disables and that there is a Pre-Arm Check error. How can I view what is causing this error? I have tried accessing the flight logs but that does not seem to provide any information. https://logs.px4.io/plot_app?log=02ade54d-d83b-41c1-b304-6ce31e9ff680
> > > > > > Any help would be much appreciated!
> > > > > 
> > > > > 
> > > > > Because kakuteh7 does not have a mag, it is necessary to set ekf to disable the mag and use the external heading angle for positioning.
> > > > 
> > > > 
> > > > I have disabled that. With it disabled, will it never be able to arm unless I get an external heading angle? I am planning on using optitrack motion capture software and that is what I used with the PixRacer instead of GPS. Is there any information that would show up in the pre-flight checks showing the lack of any sensors?
> > > > Edit: Further reading shows that a compass is necessary. How difficult would it be to add a magnetometer? I have some generic ones lying around.
> > > 
> > > 
> > > Can you arm in stabilize mode using ekf2?I use it with motion capture too. You can use yaw information from motion capture instead of mag.
> > 
> > 
> > I cannot currently, probably due to lack of magnetometer. I need to connect an esp8266-01 wifi module for the motion capture information. Unless there is an easier way to get mocap to the board. I know how to get the yaw information and set up everything for motion capture and autonomous flights, I hope. I did this before with a PixRacer and had tremendous success, just the drone was rather heavy and not able to hold the Z position. I am glad that someone has gotten this working! Hopefully I can get everything working for a demonstration I have to give Thursday!
> 
> To get the information of motion capture,I use a offboard computer to receive the information and use mavros via telem2.To use telem2,you should change the compile file.The file is https://github.com/PX4/PX4-Autopilot/blob/main/boards/holybro/kakuteh7/default.px4board.(Because it doesn't have /dev/ttyS1.)If you don't use gps,you can use telem2 in /dev/ttyS3.

DEADLYBBS, 2022-11-15:
@ckl677 I think I can get away without modifying the files. My solution is to use Telem1 by taking the Rx and Tx lines from the JST-GHR-6P port. I will let you know how this goes.

DEADLYBBS, 2022-11-15:
> @ckl677 I think I can get away without modifying the files. My solution is to use Telem1 by taking the Rx and Tx lines from the JST-GHR-6P port. I will let you know how this goes.

I have gotten the wifi module wired correctly, tx to rx and rx to tx using the telem1 serial port, the baud is also set correctly (I have it set to 921600 on QGC and the ESP8266 board). Just it will not connect to QGC. MAV_0_CONFIG is set to TELEM 1. The rest are disabled. Cannot figure out what is wrong with it. My cport and hport are correct as well.

DEADLYBBS, 2022-11-15:
> I have gotten the wifi module wired correctly, tx to rx and rx to tx using the telem1 serial port, the baud is also set correctly (I have it set to 921600 on QGC and the ESP8266 board). Just it will [...]

I have tried using GPS 1 by disabling the gps and setting mavlink to use GPS. Still nothing.

DEADLYBBS, 2022-11-16:
> > > > @ckl677 I think I can get away without modifying the files. My solution is to use Telem1 by taking the Rx and Tx lines from the JST-GHR-6P port. I will let you know how this goes.
> > > 
> > > 
> > > I have gotten the wifi module wired correctly, tx to rx and rx to tx using the telem1 serial port, the baud is also set correctly (I have it set to 921600 on QGC and the ESP8266 board). Just it will not connect to QGC. MAV_0_CONFIG is set to TELEM 1. The rest are disabled. Cannot figure out what is wrong with it. My cport and hport are correct as well.
> > 
> > 
> > I have tried using GPS 1 by disabling the gps and setting mavlink to use GPS. Still nothing.
> 
> Can you get the local position msg in qgc?

Turns out, the daily build for QGC cannot see the wifi from the esp8266 chip.

DEADLYBBS, 2022-11-16:
Is bi-directional dshot supported for this board?

shupx, 2022-11-16:
@DEADLYBBS The dshot is fine with the old kakute h7 with mpu6000. I have tested it yesterday

DEADLYBBS, 2022-11-16:
@shupx but is it bidirectional? I am having a weird issue where my drone does not want to stay level in stabilized flight mode. Instead is kinda lightly bounces. Didn't know if dshot would help. I am using Dshot 1200 currently.

shupx, 2022-11-16:
@ckl677 How did you modify the compile file to use telem2? The uart2 is used by the bluetooth and it seems that there is no pad for telem2 on the kakute h7 board.
[image]

w407022008, 2022-12-15:
@ckl677 Hello, I didn't use TELE2 yet, so I can't be sure. Can you tell me if I can use it after disabling the ESP32 C3 bluetooth chip? A pull-up resistor of 10kΩ is marked in the figure. You can use a flywire to pull it down.

[image]

junwoo091400, 2023-03-01:
Discussion seems to have diverged into TELEM port configuration & DSHOT. Closing as resolved

closed 2023-03-01 by junwoo091400