PX4/PX4-Autopilot #19995: Timestamp format of `mavlink_hil_optical_flow_t` not properly handled

obicons, 2022-07-30:
## Describe the bug
In [handle_message_hil_optical_flow](https://github.com/PX4/PX4-Autopilot/blob/8cc6d02af324ba713beb80b374236f70ac7f0a9a/src/modules/mavlink/mavlink_receiver.cpp#L856), the `sensor_optical_flow` timestamp is assigned `flow.integration_time_us`. But `flow.integration_time_us` could be in any format -- time since system boot or UTC. The representation should be canonicalized (e.g., to time since system boot).

## To Reproduce
Mix and match timestamps sent to `handle_message_hil_optical_flow`.

## Expected behavior
The timestamps should be canonicalized.

dagar, 2022-07-30:
It's the flow integration time (eg time between frames) in microseconds.

https://github.com/PX4/PX4-Autopilot/blob/8cc6d02af324ba713beb80b374236f70ac7f0a9a/src/modules/mavlink/mavlink_receiver.cpp#L856

The actual timestamp is set to the current system time, but probably should be updated to use the time sync (if available).

https://github.com/PX4/PX4-Autopilot/blob/8cc6d02af324ba713beb80b374236f70ac7f0a9a/src/modules/mavlink/mavlink_receiver.cpp#L850

obicons, 2022-07-31:
Thanks for consideration, @dagar! I'm happy to create a PR with the change. To be sure, how to access the time sync?