PX4/PX4-Autopilot #19998: Level Horizon in QGC not working with main or developer v1.13

benberry3d, 2022-08-02:
## Describe the bug
In QGroundControl (daily build), Level Horizon calibration button does nothing.  No error messages and sens_board_x_off or sens_board_y_off are not updated.

## To Reproduce
1. Flash either "Developer" branch option in QGC or build and flash Main branch onto a Pixhawk 4 or Pixhawk 6X (have tried both)
2. Attempt to use Level Horizon calibration tool under Sensors in QGC
3. Dialog will pop up to start calibration, click OK
4. After clicking OK, dialog box disappears, and there is no other message or change to indicate anything happened.
5. Verify that SENS_BOARD_X_OFF and SENS_BOARD_Y_OFF parameters have not changed

## Expected behavior
Expect indications in GUI of carrying out the calibration (either success or failure), and that the parameters mentioned above get updated (if board is not perfectly level already, which is was not)

## Log Files and Screenshots
Log file from the Developer branch tested
https://review.px4.io/plot_app?log=a8e7148c-d405-41a8-9247-2cb3d288deb1
branch: https://github.com/PX4/PX4-Autopilot/commit/c5f72fb5d9dd296b7e77c184c94540a0293ae3c2

(also pulled main from Aug 1 and same issue)

## Additional context
With v1.13.0 Stable or Beta (https://github.com/PX4/PX4-Autopilot/commit/6823cbc4140e29568f00e1211ae60e057adb1a1f), the Level Horizon button works as expected.

tstastny, 2022-08-17:
thanks @benberry3d for reporting the issue! - I also just noticed this myself.

I don't have a spare moment to dive in deeper today or tomorrow - but after a (very quick / un-thorough) browse through the code, it seems that *if* the calibration routine was actually called, a message should be output regardless:
https://github.com/PX4/PX4-Autopilot/blob/deb938fceaa79b927ae0f622d8b2b8c4ad10d391/src/modules/commander/level_calibration.cpp#L176-L185

Since I didn't see any message outputting a failure or success, I suppose this means it could be a QGC issue? (not sending the command)? @dagar @bresch -- quick thoughts if I overlooked something (very possible)? Otherwise.. I will try to resume looking into it on Friday.

On a separate, less important, note: I also noticed the rotational offsets from this routine are only displayed if they are greater than 0.1 degrees. It's not critical... but... why? why not just show what the calibration did always?
https://github.com/PX4/PX4-Autopilot/blob/deb938fceaa79b927ae0f622d8b2b8c4ad10d391/src/modules/commander/level_calibration.cpp#L160-L166

vincentpoont2, 2022-08-23:
Looks like this issue is still there at this moment with the Main Branch. No issue with Release/1.13 Branch. 
Tested with Pixhawk 6X & 6C & QGC Daily. @dagar @bresch

benberry3d, 2022-08-24:
It looks like this commit (https://github.com/PX4/PX4-Autopilot/pull/19729) modified/deleted the if statement that starts the level calibration in commander.cpp.  Can test on hardware tomorrow.  @junwoo091400 was this change intentional?

Before (see line 1462):
https://github.com/PX4/PX4-Autopilot/blob/af4038aa7e227833c15bd5b5297caf0ef8c4465d/src/modules/commander/Commander.cpp#L1456-L1473

After:
https://github.com/PX4/PX4-Autopilot/blob/32ae00fd44107560c6b2ef5bfca58e640fb145de/src/modules/commander/Commander.cpp#L1456-L1472

benberry3d, 2022-08-26:
Confirmed the level horizon functionality broke between those two commits.  I'll try adding the else if statement back in on main and see if that fixes it.  Also observed that sens_board_x_off  (and y) round to nearest integer in QGC now.  Fairly sure that used to have one decimal place accuracy.  A separate issue, though.

junwoo091400, 2022-08-27:
Thanks for the report. I can definitely confirm that my PR accidentally contained this modification which was unintentional!

We're you able to create the fix commit?

benberry3d, 2022-08-27:
Yeah I just reverted that section on main branch but my testing is hitting some snags with a pixhawk 6c, when I load the firmware the attitude estimator is not running!  (Though it did at least start the level calibration routine).  Loading the main branch without my changes to see if the issue still remains.  With 1.13 stable release the attitude estimator works fine.  So hopefully I can get this fix in once I figure out these issues and test that it completes a level calibration.

benberry3d, 2022-08-27:
Ok yeah seems like some issue with main branch and the Pixhawk 6C.... switched to a Pixhawk 4 and no issues with the attitude estimator.  Level horizon now works with the change.  I'll submit a PR and tag you.  Thanks!

benberry3d, 2022-08-27:
PR submitted: https://github.com/PX4/PX4-Autopilot/pull/20117

junwoo091400, 2022-08-29:
> attitude estimator is not running!

Could it be the case that you have the multi-EKF enabled, and due to the wrong settings none of the EKF2 instance is selected, hence attitude estimator not running (EKF2) in the end?

Check out this doc for the parameters that you need to check: https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#running-a-single-ekf-instance

benberry3d, 2022-08-29:
I will try that, thanks!  When it wasn't working for me the highres_imu data was also missing / zero for the mag and barometer fields for the 6C.  If I still see a problem I will raise a separate Issue.  In the meantime I will close this.

closed 2022-08-29 by benberry3d