PX4/PX4-Autopilot #20007: Error with CRYPTO set: fatal error: crypto_backend_definitions.h: No such file or directory

TByte007, 2022-08-04:
I added CRYPTO to a custom config and I get this error:

```
In file included from /home/tbyte/pilots/PX4-Autopilot/src/modules/logger/log_writer_file.h:42,
                 from /home/tbyte/pilots/PX4-Autopilot/src/modules/logger/log_writer.h:36,
                 from /home/tbyte/pilots/PX4-Autopilot/src/modules/logger/logger.h:36,
                 from /home/tbyte/pilots/PX4-Autopilot/src/modules/logger/logger.cpp:37:
/home/tbyte/pilots/PX4-Autopilot/platforms/common/include/px4_platform_common/crypto.h:44:10: fatal error: crypto_backend_definitions.h: No such file or directory
   44 | #include "crypto_backend_definitions.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```
I can see that the file is in `src/drivers/sw_crypto/crypto_backend_definitions.h` but I dont understands why it is not included with the CRTYPTO setting ?

Sagar-Anigere, 2025-04-17:
is this fixed?

Sagar-Anigere, 2025-04-17:
fixed -> copy the contents of cryptotest.px4board to default.px4board(instead of make seprate file this will fix your issue)